raspiblitz/raspiblitz #4324: [RP5] NVMe/LaptopSSD Boot Support

rootzoll, 2023-12-16:
This is a bigger task to develop because it should be the same process basically with reusing a Laptop with a single SSD. To consider are:

- Initial Setup with sdcard/bootableusbdrive
- Changing defaultboot to SSD/NVMe
- Using sdcard/usbdrive for emergeny backups
- When sdcard/usbdrive with fresh image gest connected/booted it triggers recover/update of system

For background on NVMe boot .. see #4259

rootzoll, 2024-02-16:
Also with this change (migration to boot from SSD/NVMe) its to consider to recommend 2TB nvme/ssd as default.

tysecure, 2024-04-15:
In case any one was curious:

Raspberry Pi 5, 8GB: Indexing Fulcrum from 0 to 100%

Time to Completion:
2TB NVMe: 23hrs 🔥
2TB SSD via USB 3.0: 5 days

[image]
[image]

rootzoll, 2024-10-29:
Check: https://www.tomshardware.com/raspberry-pi/how-to-install-and-use-the-new-raspberry-pi-boot-menu

rootzoll, 2024-12-06:
Some notes on how to change the boot order by command:

```
sudo rpi-eeprom-config --out bootconf.txt
set to BOOT_ORDER=0xf461 in bootconf.txt
sudo rpi-eeprom-config --apply bootconf.txt
sudo reboot
```

`0xf461` means it will do sd-card first, then NVMe, then USB-Massstorage, the reboot

To render the sd card non-bootable the remove of the boot partition seems to work:
```
sudo umount /dev/mmcblk0p1
sudo parted --script /dev/mmcblk0 rm 1
reboot
```

rootzoll, 2024-12-08:
TODOs:

- [ ] new `blitz.data.sh` to step by step replace `blitz.datadrive.sh`
- [ ] detect if running raspberrypi, laptop, VM
- [ ] detect if running on raspberrypi <5 
- [ ] detect if PCIe/SATA or USB connected drive
- [ ] detect all possible data drives with their sizes  
- [ ] recognize when user flashed image wrongly to NVMe
  - feedback to user SSH/Web --> image flash to 64GB sd card
  - check/change boot order