raspiblitz/raspiblitz #4346: Fulcrum error on RPI5

openoms, 2024-01-01:
```
Jan 01 18:21:03 raspberrypi systemd[1]: Started fulcrum.service - Fulcrum.
Jan 01 18:21:03 raspberrypi Fulcrum[1301304]: <jemalloc>: Unsupported system page size
Jan 01 18:21:03 raspberrypi Fulcrum[1301304]: <jemalloc>: Unsupported system page size
Jan 01 18:21:03 raspberrypi Fulcrum[1301304]: <jemalloc>: Unsupported system page size
Jan 01 18:21:03 raspberrypi Fulcrum[1301304]: terminate called without an active exception
Jan 01 18:21:03 raspberrypi systemd[1]: fulcrum.service: Main process exited, code=killed, status=6/ABRT
Jan 01 18:21:03 raspberrypi systemd[1]: fulcrum.service: Failed with result 'signal'.
```

Solution:
https://t.me/raspibolt/23047
>Fix for Fulcrum on Pi5:
Add the following to /boot/config.txt
kernel=kernel8.img
This will change page size to 4k from 16k

neversleep404, 2024-01-02:
I have a Pi 5, but I'm not familiar with the problem. Which module is encountering the issue? What module use Fulcrum? I'm running raspiblitz-arm64-rpi-image-2023-12-15-40587e4 with Kernel 6.1.0-rpi7-rpi-2712 and had to start the compilation myself for some modules like CLBOSS. Both Blitz API and WebUI aren't functioning and were generating a very high load (load of 5), but everything runs fine without them (load of 1.5).

openoms, 2024-01-03:
This exposes a more widespread issue with a growing list of software not compatible with the 16k byte PAGE_SIZE: https://github.com/raspberrypi/bookworm-feedback/issues/107

I suggest we should use the `kernel8.img` by default.

As it is a kernel change it should be part of the `build_sdcard.sh`.

neversleep404, 2024-01-03:
> I suggest we should use the `kernel8.img` by default.
> 
> As it is a kernel change it should be part of the `build_sdcard.sh`.

II also recommend defaulting to 'kernel8.img'. Switching from a 4K to a 16K page size kernel on a Raspberry Pi 5 yields a 7% performance boost in random memory access, but be aware of potential issues, especially for non-standard page sizes. This performance gain is more pronounced in intensive computing tasks, though it may not be noticeable in everyday use.

https://forums.raspberrypi.com/viewtopic.php?p=2172765#:~:text=The%20default%20kernel%20for%20a,increase%20in%20random%20memory%20accesses

neversleep404, 2024-01-03:
Interestingly, I have noticed that the load on Kernel 6.1.0-rpi7-rpi-v8 under raspiblitz-arm64-rpi-image-2024-01-03-35d0c22 is 10-20% lower than on raspiblitz-arm64-rpi-image-2023-12-15-40587e4 with Kernel 6.1.0-rpi7-rpi-2712.
(Running CORE Lightning, CL BOSS, BTC Mempool Space, RTL. Debug for TOR and I2P off and maxuploadtarget=500000)

openoms, 2024-01-06:
agree that it could be interesting to do some benchmarking, but until the list of incompatible applications is so long it does not worth it to have the nonstandard 16K page size as default.

neversleep404, 2024-01-07:
agree. The selection of 4K paging for operations brings many benefits, and for now, it's not beneficial to consider 16K paging.

rootzoll, 2024-03-19:
Closing for release.

closed 2024-03-19 by rootzoll