raspiblitz/raspiblitz #4352: Update to mempool 2.5.0

rootzoll, 2024-01-05:
https://github.com/mempool/mempool/releases/tag/v2.5.0. 

@openoms were there any problems in the past with updating to this version?

openoms, 2024-01-06:
on Debian12 amd64 getting so there needs to be more changes than the verison number:
```
.setMaxListeners() to increase limit
(node:2043387) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [TLSSocket]. Use emitter.setMaxListeners() to increase limit
(node:2043387) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 close listeners added to [TLSSocket]. Use emitter.setMaxListeners() to increase limit
npm WARN deprecated @npmcli/move-file@2.0.1: This functionality has been moved to @npmcli/fs
npm WARN deprecated querystring@0.2.0: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.
npm WARN deprecated popper.js@1.16.1: You can find the new Popper v2 at @popperjs/core, this package is dedicated to the legacy v1
^[(##################) ⠇ reify:typescript: http fetch GET 200 https://registry.npmjs.org/typescript/-/typescript-4.6.4.tgz 54158ms (cache mi

added 1258 packages, and audited 1259 packages in 1m

177 packages are looking for funding
  run `npm fund` for details

14 vulnerabilities (7 moderate, 4 high, 3 critical)

To address issues that do not require attention, run:
  npm audit fix

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.

> mempool-frontend@2.5.0 build
> npm run generate-config && npm run ng -- build --configuration production --localize && npm run sync-assets && npm run build-mempool.js

> mempool-frontend@2.5.0 generate-config
> node generate-config.js

mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.5.0
mempool revision 57eddac7f
src/resources/config.js file not found, creating new config file
CONFIG:  (function (window) {
  window.__env = window.__env || {};
    window.__env.GIT_COMMIT_HASH = '57eddac7f';
    window.__env.PACKAGE_JSON_VERSION = '2.5.0';
  }(this));
src/resources/config.js file saved

> mempool-frontend@2.5.0 ng
> ./node_modules/@angular/cli/bin/ng.js build --configuration production --localize

✔ Browser application bundle generation complete.
An unhandled exception occurred: The service is no longer running
See "/tmp/ng-XlB8xL/angular-errors.log" for further details.
```

Der-Zett, 2024-01-06:
I have manually installed mempool version 2.5.0. The application has worked without any problems so far.

My environment
RaspiBlitz v1.11.0rc1 on PI5 8 GB RAM

I have done the following:

Change value of variable pinnedVersion="v2.4.0" to pinnedVersion="v2.5.0" in script /home/admin/config.scripts/bonus.mempool.sh

But an error occurred during the install (see below).

I have cleaned my system and listed all the steps in the script manually.

sudo adduser --system --group --home /home/mempool mempool
cd /home/mempool
sudo -u mempool git clone https://github.com/mempool/mempool.git
cd mempool
sudo -u mempool git reset --hard v2.5.0
sudo -u mempool /home/admin/config.scripts/blitz.git-verify.sh "wiz" "https://github.com/wiz.gpg" “A394E332255A6173"
cd frontend
sudo -u mempool NG_CLI_ANALYTICS=false npm install --no-optional
sudo -u mempool NG_CLI_ANALYTICS=false npm run build

***SNIP—ERROR***
> mempool-frontend@2.5.0 build
> npm run generate-config && npm run ng -- build --configuration production --localize && npm run sync-assets && npm run build-mempool.js
> mempool-frontend@2.5.0 generate-config
> node generate-config.js
mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.5.0
mempool revision 57eddac7f
src/resources/config.js file not found, creating new config file
CONFIG:  (function (window) {
  window.__env = window.__env || {};
    window.__env.GIT_COMMIT_HASH = '57eddac7f';
    window.__env.PACKAGE_JSON_VERSION = '2.5.0';
  }(this));
src/resources/config.js file saved
> mempool-frontend@2.5.0 ng
> ./node_modules/@angular/cli/bin/ng.js build --configuration production --localize

✔ Browser application bundle generation complete.
An unhandled exception occurred: The service is no longer running
See "/tmp/ng-2Xe8N6/angular-errors.log" for further details.
***SNAP—ERROR*** (Same error as when running the script)

I HAVE MAKE THIS
sudo -u mempool NG_CLI_ANALYTICS=false npm audit fix --force

AND REPEAT THIS AND IT WORKS
sudo -u mempool NG_CLI_ANALYTICS=false npm run build

I MAKE THE SAME PROCEDURE FOR DIR BACKEND
sudo -u mempool NG_CLI_ANALYTICS=false npm install --no-optional
sudo -u mempool NG_CLI_ANALYTICS=false npm audit fix --force
sudo -u mempool NG_CLI_ANALYTICS=false npm run build

VERIFY
compgen -u | grep -c mempool
1

sudo /home/admin/config.scripts/bonus.mempool.sh on

***SNIP***
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
*** Install mempool systemd ***
Created symlink /etc/systemd/system/multi-user.target.wants/mempool.service → /etc/systemd/system/mempool.service.
# OK - the mempool service is now enabled
# OK - the mempool.service is enabled, system is on ready so starting service
# needs to finish creating txindex to be functional
# monitor with: sudo tail -n 20 -f /mnt/hdd/bitcoin/debug.log

# Hidden Service for mempool
HiddenServiceDir /mnt/hdd/tor/mempool
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:4082
HiddenServicePort 443 127.0.0.1:4083

Reloading Tor to activate the Hidden Service...

The Tor Hidden Service address for mempool is:
xxxxxxx.onion
use with the port: 80
or the port: 443
result=‚OK'
***SNAP***

sudo /home/admin/config.scripts/bonus.mempool.sh status

version='v2.5.0'
codebase=1
configured=1
installed=1
localIP='XXX.XXX.XXX.XXX'
httpPort='4080'
httpsPort='4081'
httpsForced='0'
httpsSelfsigned='1'
authMethod='none'
fingerprint=‚XX:XX:XX…..‘
toraddress=‚xxxxxxxxx.onion'
isIndexed=1
indexInfo=‚'

sudo systemctl status mempool
● mempool.service - mempool
     Loaded: loaded (/etc/systemd/system/mempool.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-01-06 14:00:55 GMT; 4min 43s ago
   Main PID: 2378630 (node)
      Tasks: 22 (limit: 9569)
        CPU: 1min 13.896s
     CGroup: /system.slice/mempool.service
             └─2378630 /usr/bin/node --max-old-space-size=2048 dist/index.js

If I start the script with the parameter "off" and then with "uninstall", everything runs without any problems.

openoms, 2024-01-06:
the problem is that this error:
```
✔ Browser application bundle generation complete.
An unhandled exception occurred: The service is no longer running
See "/tmp/ng-XlB8xL/angular-errors.log" for further details.
```
breaks the script and needs interaction.

Could use `timeout`, but would really need a better solution.

Der-Zett, 2024-01-07:
Maybe 

```
pnpm install %70 faster
or
npm install --no-audit 15% faster
```

or define timeouts in ~/.npmrc

```
// npm config ls -l
// add these 2 lines in ~/.npmrc
fetch-retry-maxtimeout = 6000000
fetch-retry-mintimeout = 1000000
```

openoms, 2024-01-07:
> Maybe
> 
> ```
> pnpm install %70 faster
> or
> npm install --no-audit 15% faster
> ```
> 
> or define timeouts in ~/.npmrc
> 
> ```
> // npm config ls -l
> // add these 2 lines in ~/.npmrc
> fetch-retry-maxtimeout = 6000000
> fetch-retry-mintimeout = 1000000
> ```

these don't help as the breaking error is when running:
```
> ./node_modules/@angular/cli/bin/ng.js build --configuration production --localize
```
not with audit, also using `npn audit fix` in the script doesn't help.
I can make it finish manually too after exiting the error, but need to find the way for the automated script to run through.

Best would be to find a solution to the error.

Der-Zett, 2024-01-07:
I have ask in Mempool-Support. But I don't have an answer yet.

Many who receive the same error messages indicate that the following modules should be updated
nodejs
npm
angular
I have updated everything but the error remains.

Der-Zett, 2024-01-07:
First response from Mempool-Support:

"To me it seems your hardware simply does not not have enough resources to build all localized bundles? Maybe try to only localize in English see if that works for you."

openoms, 2024-01-13:
Tried removing all localization, but getting the same error. I am trying to compile o a laptop with 16GB memory so resources shouldn't be the issue.

```
  cd frontend || exit 1

  jq '.projects.mempool.i18n.locales = {}' angular.json | sudo -u mempool tee angular_modified.json

  # Replace the original angular.json with the modified one
  sudo -u mempool  mv angular.json angular_backup.json
  sudo -u mempool  mv angular_modified.json angular.json
```

I wonder that why @Emzy 's instance is also on v2.4.0 https://mempool.emzy.de/about ?
Did you have any problem compiling?

rootzoll, 2024-01-18:
Worked on my test - ready for next release candidate.

rootzoll, 2024-03-19:
Restested OK. Closing for release.

closed 2024-03-19 by rootzoll