raspiblitz/raspiblitz #4354: I can received sats on-chain but cannot do/send any transactions On-Chain

chamoBtC, 2024-01-07:
**Describe the bug**
Not sure if this is a bug but for some strange reason I cannot send any transactions through On-Chain  I  tried few times with no success. Always is  messsage of Error. Within the browser interface sometimes I cannot see my balance or transactions and yet I can see my balance in Ride the Lightning.

See attachment 

Please advise

Thank you 

[image]

[image]

[image]

chamoBtC, 2024-01-07:
See balance zero in the browser interface

[image]

rootzoll, 2024-01-18:
Can you see that the transaction you send has been confirmed on the blockchain?
Have you alraedy tried to send a transaction from your node and that one might still wauting for confirmation?

chamoBtC, 2024-01-18:
Will send you a screenshot later. Still can't send any sats through the node. I have almost like 1M sats and now it says that I only have 906 sats. Not sure if it is because I changed the name of my node. Very confused

chamoBtC, 2024-01-22:
here is the message I  get enytime I want to send sats on chain

[image]

rootzoll, 2024-02-09:
It looks like you've encountered an error while trying to broadcast a transaction on the blockchain. It simply means that the fee that has been set for the transaction is lesser than the minimum fee that has to be paid for a transaction to be accepted into the network, into the memory pool (mempool), from where the transaction will sit waiting to be confirmed.