raspiblitz/raspiblitz #4356: Failed to start ...n daemon on mainet after micro sd update

Damian101101, 2024-01-09:
**Describe the bug**
Have just updated to v1.10.0 and after entering a new password A, I get this screen:

[image]

Then the screen keeps looping

**To Reproduce**
Restart with freshly flashed =micro sd card

**Debug file**

If possible, please add a debug report. To do that, drag a text file with the debug report in here.

rootzoll, 2024-01-18:
This looks like the test run of the bitcoind during recover/Provision is not working. I will mark this to take a look for v1.11.1 if there is anything during that test routine to better detect a failed start.

To see why it has problems I would need a debug report: ssh into the device. Use CTRL+c to get to the terminal and run the command `debug` - post result here.

rootzoll, 2024-09-27:
OK normally a failed testrun of bitcoind should be detected here and if it fails it should exit with an error:
https://github.com/raspiblitz/raspiblitz/blob/efacac08480844b044323bab137aa59d95b84444/home.admin/_provision.setup.sh#L140-L141

The "Setup System .." message happens here - which is displayed on the LCD:
https://github.com/raspiblitz/raspiblitz/blob/efacac08480844b044323bab137aa59d95b84444/home.admin/_provision_.sh#L113-L114

So the error message on the LCD comes after `_provision.setup.sh` with its testrun - which is a bit strange. So bitcoind startup - delivers a valid `getblockchaininfo` report on testrun BUT THEN seems to fail afterwards.

I keep this issue open - if anybody else experience this error, please CTRL+c to get to the terminal and run the command debug - post result here. Without further debug info I cannot tell why this edge case occurs.