raspiblitz/raspiblitz #4359: [RPi5] Support Pimoroni NVMe Base

rootzoll, 2024-01-11:
https://shop.pimoroni.com/products/nvme-base?variant=41219587178579
https://www.youtube.com/watch?v=odG7FbptgWQ
Should already work from the configs with the adaption of pineberrypi nvme board ('pciex1_gen=3' not needed) - BUT ...

it needs at least the raspberryPi has an updated firmeware `Wed  6 Dec 18:29:25 UTC 2023 (1701887365)` which did not come with the first batch of RPi5s

To see installed version:
`sudo rpi-eeprom-update`

To install new firmware (manual trigger of reboot needed after that):
`sudo rpi-eeprom-update -a`

redmac68, 2024-01-11:
I can confirm it works with raspiblitz v1.11.0rc1 (basic setup - bitcoin, LND, RTL running)
Definitely needs the firmware update as above first or the nvme disk is not detected.
I am running with 'pciex1_gen=3' on a test rpi5 8GB with Samsung_SSD_970_EVO_Plus_2TB_S6P1NS0T208608F. 
Box is doing no useful work - no sats, no channels.

neversleep404, 2024-01-12:
How would the migration process work?
https://github.com/raspiblitz/raspiblitz/issues/4259#issuecomment-1880322366

rootzoll, 2024-01-15:
Please add all the NVMes (manufacture & model & size) that you are running with the PIMORONI Base as a comment to this issue - for performance test you can use: `sudo hdparm -t --direct /dev/nvme0n1p1`

TrezorHannes, 2024-01-15:
Using Pimoroni NVMe Base with OS on SD (allows for quicker update iterations).

## NVMe
```
Crucial P3 2TB PCIe M.2 2280 SSD
CT2000P3SSD8
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            2342E880D515         CT2000P3SSD8                             1           2.00  TB /   2.00  TB    512   B +  0 B   P9CR30A
```

```
sudo hdparm -t --direct /dev/nvme0n1p1

/dev/nvme0n1p1:
 Timing O_DIRECT disk reads: 1236 MB in  3.00 seconds = 411.38 MB/sec
 ```

david-bakin-sl, 2024-01-16:
I'm about to build a RPi5+Pimoroni NVMe Base+2TB NVMe ... and was wondering if I dare try to sync the whole chain (rather than start with a copy of the chain from my existing 'blitz).  Would the fast drive be expected to improve sync-from-genesis time?  (And if so, how much?)

redmac68, 2024-01-16:
In my case full sync was much faster ~ 4 days.

neversleep404, 2024-01-16:
A copy of the blockchain from the USB SSD directly to the NVMe should be much faster. With a blockchain size of 550GB and a conservatively estimated USB transfer rate of 400 MB/s, the copying process only takes 24 minutes.
Copying from one Raspberry Pi to another takes several hours, and a full sync over the internet takes about 3-6 days (I have a gigabit internet connection) and depends on the other Bitcoin full nodes. Lightning and Electrum also take their time.

rootzoll, 2024-01-17:
@neversleep404 
> A copy of the blockchain from the USB SSD directly to the NVMe should be much faster.

Yeah it seems to make sense to provide a migration tools now to copy from external HDD/SSD. I would target that for v1.11.1 to not further delay the v1.11.0 release. Will opened a new ticket for that: #4377

Can somebody in the meantime test if on the RaspberryPi5 with the Pimoroni NVMe Base an additional connected USB SSD (the one we used before) can still copy lots of dummy data correctly without drops in power?

PR #4365 (2024-01-13): Updating Firmware if needed for RaspberryPi5 to get Pimoroni NVMe Base working #4359
fixes #4359
closed 2024-03-19 by rootzoll