raspiblitz/raspiblitz #4364: CLN Developer Mode crashes Lightningd

sninenwi22, 2024-01-13:
Thanks for your help. I'm running a Raspblitz with CLN. I'm trying to dev-forget a channel when I try it I get the following error:

```
 lightning-cli dev-forget-channel{
   "code": -32601,
   "message": "Unknown command 'dev-forget-channel'"
}
```

I thought if would work if I enabled developer mode. However, when I edit the config file to run in developer mode it fails to start.

Here is the config file excerpt:
```
....
[Service]
ExecStartPre=-/home/admin/config.scripts/cl.check.sh prestart mainnet
#ExecStart=/bin/sh -c '/usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --pid-file=/run/lightningd/lightningd.pid --rpc-file-mode 0660'
ExecStart=/bin/sh -c '/usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --developer --pid-file=/run/lightningd/lightningd.pid --rpc-file-mode 0660'
ExecStartPost=-/home/admin/config.scripts/cl.check.sh poststart mainnet
....

```
Restart daemon
`- sudo systemctl daemon-reload`

Then run
`- sudo systemctl restart lightningd`

I get this error:
```
sudo systemctl restart lightningd
Job for lightningd.service failed because the control process exited with error code.
See "systemctl status lightningd.service" and "journalctl -xe" for details.
```

Here is the systemctl status lightningd.service output
```
Loaded: loaded (/etc/systemd/system/lightningd.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Fri 2024-01-12 16:57:10 GMT; 28s ago
Process: 368686 ExecStartPre=/home/admin/config.scripts/cl.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
Process: 368712 ExecStart=/bin/sh -c /usr/local/bin/lightningd --developer --conf=/home/bitcoin/.lightning/config   (code=exited, status=1/FAILURE)
Process: 368716 ExecStartPost=/home/admin/config.scripts/cl.check.sh poststart mainnet (code=exited, status=0/SUCCESS)
Main PID: 368712 (code=exited, status=1/FAILURE)
CPU: 1.244s
```

I've also tried to edit the config file by putting in the "developer" flag like below:
```
developer
network=bitcoin
log-file=cl.log
log-level=info
plugin-dir=/home/bitcoin/cl-plugins-enabled
```

On restart through doing this I get the same issue. Any ideas? Thanks in advance!

openoms, 2024-01-13:
this is not a bug but a misconfiguration. 

Did you remove the `--developer` from systemd when trying it in the config?

do you have link to the CLN docs ormpart of the manpage where this is described?

sninenwi22, 2024-01-13:
Hi I tried both the "--developer" flag in systemd and "developer" in the config independently. 

Yes, here is the link to CLN docs = https://docs.corelightning.org/reference/lightningd-config

I set back to default install just now and just did the "developer" flag. Following the steps below but still getting the error. I'm wondering if maybe the CLN can be compiled without the developer mode support? 

Here's how to reproduce the issue following the docs of CLN failing on Raspiblitz

1. update the config file following the docs with the developer mode flag:
```

network=bitcoin
log-file=cl.log
log-level=info
plugin-dir=/home/bitcoin/cl-plugins-enabled

# Enable developer mode
developer

# Tor settings
proxy=127.0.0.1:9050
bind-addr=127.0.0.1:9736
addr=statictor:127.0.0.1:9051/torport=9736
always-use-proxy=true

rpc-file-mode=0660
```

2. After updating the configuration file, restart the lightningd service to apply the changes:

Reload the systemd daemon configuration:

`sudo systemctl daemon-reload`

3. Restart the lightningd service:

`sudo systemctl restart lightningd`

4. Service fails to start on running the logs command:
`sudo journalctl -u lightningd`

Here is the output (see developer) unknown issue:

```
Jan 13 13:46:45 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Jan 13 13:46:45 raspberrypi bitcoin[3868410]: srw-rw---- 1 bitcoin bitcoin 0 Jan 12 19:45 /mnt/hdd/app-data/.lightning/bitcoin/lightning-rpc
Jan 13 13:46:45 raspberrypi systemd[1]: lightningd.service: Can't open PID file /run/lightningd/lightningd.pid (yet?) after start-post: Operation not permitted
Jan 13 13:46:48 raspberrypi sh[3868405]: lightningd: Config file /home/bitcoin/.lightning/config line 9: developer: unknown option
Jan 13 13:46:48 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 13:46:48 raspberrypi systemd[1]: lightningd.service: Daemon never wrote its PID file. Failing.
Jan 13 13:46:48 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Jan 13 13:46:48 raspberrypi systemd[1]: Failed to start c-lightning daemon on mainnet.
Jan 13 13:46:48 raspberrypi systemd[1]: lightningd.service: Consumed 1.254s CPU time.
```

openoms, 2024-01-13:
Which version are you using?
The developer option works on my node running CLN v23.11.2

sninenwi22, 2024-01-13:
Thanks will try and come back.

sninenwi22, 2024-01-14:
I was running v23.08.1 which came with the  v1.10.0. Now I've upgraded to CLN v23.11.2 but the command is still not accessible. How are you putting it in developer mode?

```
lightning-cli dev-forget-channel
{
   "code": -32601,
   "message": "Command \"dev-forget-channel\" is developer-only"
}
```

When I put the flag "developer" in the config file I'm getting a segfault:

/home/bitcoin/.lightning/config
```
#lightningd configuration for bitcoin mainnet

developer

network=bitcoin
log-file=cl.log
log-level=info
plugin-dir=/home/bitcoin/cl-plugins-enabled

# Tor settings
proxy=127.0.0.1:9050
bind-addr=127.0.0.1:9736
addr=statictor:127.0.0.1:9051/torport=9736
always-use-proxy=true
```

sudo journalctl -u lightningd -e
```Jan 14 22:32:06 raspberrypi systemd[1]: lightningd.service: Consumed 1min 4.301s CPU time.
Jan 14 22:32:06 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Jan 14 22:32:07 raspberrypi bitcoin[99906]: srw-rw---- 1 bitcoin bitcoin 0 Jan 14 22:21 /mnt/hdd/app-data/.lightning/bitcoin/lightning-rpc
Jan 14 22:32:07 raspberrypi systemd[1]: lightningd.service: Can't open PID file /run/lightningd/lightningd.pid (yet?) after start-post: Operation not permitted
Jan 14 22:32:10 raspberrypi sh[99899]: lightningd: FATAL SIGNAL 11 (version v23.11.2)
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578eac65f send_backtrace
Jan 14 22:32:11 raspberrypi sh[99899]:         common/daemon.c:33
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578eac6fb crashdump
Jan 14 22:32:11 raspberrypi sh[99899]:         common/daemon.c:75
Jan 14 22:32:11 raspberrypi sh[99899]: 0x7f83f987a7 ???
Jan 14 22:32:11 raspberrypi sh[99899]:         ???:0
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578e78de4 plugin_conn_finish
Jan 14 22:32:11 raspberrypi sh[99899]:         lightningd/plugin.c:875
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f6d3bf destroy_conn
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/io/poll.c:246
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f6d3eb destroy_conn_close_fd
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/io/poll.c:252
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f79a0b notify
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/tal/tal.c:246
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f79af7 del_tree
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/tal/tal.c:437
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f7a117 tal_free
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/tal/tal.c:521
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f6bb0f io_close
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/io/io.c:450
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578f6db8b io_loop
Jan 14 22:32:11 raspberrypi sh[99899]:         ccan/ccan/io/poll.c:457
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578e47fff io_loop_with_timers
Jan 14 22:32:11 raspberrypi sh[99899]:         lightningd/io_loop_with_timers.c:22
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578e7ab7f plugins_init
Jan 14 22:32:11 raspberrypi sh[99899]:         lightningd/plugin.c:1991
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578e4e23b main
Jan 14 22:32:11 raspberrypi sh[99899]:         lightningd/lightningd.c:1082
Jan 14 22:32:11 raspberrypi sh[99899]: 0x7f83b13e17 __libc_start_main
Jan 14 22:32:11 raspberrypi sh[99899]:         ../csu/libc-start.c:308
Jan 14 22:32:11 raspberrypi sh[99899]: 0x5578e23667 ???
Jan 14 22:32:11 raspberrypi sh[99899]:         ???:0
Jan 14 22:32:11 raspberrypi sh[99897]: Segmentation fault
Jan 14 22:32:11 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=139/n/a
Jan 14 22:32:11 raspberrypi systemd[1]: lightningd.service: Daemon never wrote its PID file. Failing.
Jan 14 22:32:11 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Jan 14 22:32:11 raspberrypi systemd[1]: Failed to start c-lightning daemon on mainnet.

sninenwi22, 2024-01-14:
Issue was due to having a plugin in the plugins enabled directory. Once that was removed lightningd starts with the "developer" flag in the config

sninenwi22, 2024-01-14:
closing as a configuration issue. disable plugins to use.

closed 2024-01-14 by sninenwi22