raspiblitz/raspiblitz #4366: RAID-1 - With USB Hub on Power Supply or 2Bay NAS and mount via NFS

bvolution, 2024-01-13:
Hi, let me start by saying thank you for all the work you did in this project.
In the process of securing and in light of the current costs for channel opening and closes
I want to protect my raspiblitz against failure of one SSD and instead run on RAID-1.
Unfortunately the Raspi4 will not have sufficient power to run two SSDs connected via the USB-3.0 ports.

I already own two `SanDisk SSD PLUS 1TB` as well as the `UGREEN SATA-USB 3.0 external Cover`.
Now since I want to be extra careful to not risk my funds I wanted to ask you guys as experts if either of
the two provided solutions might work:
**Idea 1:**  
Could it might already be sufficient, for this setup to just get an USB 3.0 hub with external power supply (e.g. [Atolla](https://www.amazon.de/atolla-Netzteil-SuperSpeed-Datenhub-Intelligenter-Schwarz/dp/B07P6MPXJ7/ref=sr_1_4?keywords=usb+hub+3.0+mit+netzteil&qid=1705171146&sr=8-4), 15w).
Or would the write / read speed a problem here? 

**Idea 2:**  
Buy a 2Bay NAS Cover (i.e., without the SDD/HDD) e.g. [this](https://www.notebooksbilliger.de/qnap+systems+tr+002+erweiterungsgehuse+2+bay+b+ware+823215) plug in the 2 SSDs I already own.
And mount these two SSDs as NFS / setup RAID1 for them.
The critical question here would be if it would be possible during setup of the raspiblitz?
Is there any documentation / report / blog post on this? Do you have experience with this setup?

rootzoll, 2024-01-18:
If you looking for a raid solution that works with the RaspiBlitz setup as it is I would try a USB3 2 bay hardware raid with external power: https://www.youtube.com/watch?v=LW9uCahPdVc .. because the RaspiBlitz should just see it as one drive like normal and the whole RAID1 is managed by the hardware. But stable power is still most important - so an wellm connected external power to the RAID1 enclosure is highly recommended. If you try it out please report your experience.

bvolution, 2024-01-19:
Thanks rootzoll.
I went for a mini PC 3BAY (including 2 USB SSDs) solution, and will setup proxmox server to host the RAID-1. 
The most exciting part will be the migration from my raspberry pi 4 to the new RAID-1 setup.
I am preparing a blog on my experience and how to setup inside of proxmox. Will post it here when it is finished :)

TonyGranato, 2025-01-28:
Is there any updates regarding this. My Raspiblitz ist running out of space soon, a new raid setup is what I want in the future. So my question is: What is the best raid solution (Hardware)  for raspiblitz right now?