raspiblitz/raspiblitz #4369: Problem Blockchain Sync

Rospe85, 2024-01-15:
Hi,
A started the Raspiblitz "Bitcoin only", at the beginning everything goes smooth however at approx 6% of blockchain sync the process stop and the only solution was to restart the device. I tried to restart everything from scratch but similar problem occur.

At this link you can find my debug report.

https://l.termbin.com/54t5

Thanks in advance

Rospe

rootzoll, 2024-01-18:
Somehow your Bitcoin folder on SSD gest "ReadOnly" - this can be because:

- File System Errors: The OS might switch a file system to read-only mode if it detects errors, to prevent further damage.
- Manual Mounting: The file system could have been manually mounted as read-only, either intentionally or by mistake.
- Hardware Issues: Problems with the hard drive or its connection can lead the OS to mount it as read-only.

You can use tools like fsck (File System Consistency Check) to check and repair file system errors. It's crucial to unmount the filesystem or ensure it's not in use before running fsck. Or if there is no critical data or funds on the node you can start with a fresh sd card image CANCEL the recover and select START FRESH and format the SSD again.

cstenglein, 2024-09-30:
Since this is 8 months old and no reply, I'm closing this issue.

If you still have this issue, please re-open it.

closed 2024-09-30 by cstenglein