raspiblitz/raspiblitz #4372: Setting fee sats/vbyte for closing channel miscalculates actual fee rate

abinition, 2024-01-16:
Upon closing a channel, (I) selected 50 sats/vbyte, but when transaction was viewed on mempool, it showed 24.7 sats/vbyte.  Afterwards, tried doing a "force close" which added a CPFP, this increased fee rate to 47.5 sats/vbyte.  Hope it confirms.

Devs should check the math calculation for channel close and apply the correct fee amount from what is selected on web interface.

Hard to reproduce without opening and then closing channels, which costs $$.

rootzoll, 2024-01-18:
What did you use? LND or CLN? Which WebUI or Terminal?

abinition, 2024-01-18:
It was RTL web interface.

abinition, 2024-01-18:
re-opened on https://github.com/Ride-The-Lightning/[RTL](https://github.com/Ride-The-Lightning/RTL) issue  #1345

closed 2024-01-18 by abinition