raspiblitz/raspiblitz #4375: How do I get Electrum to run?

Nerfzooka, 2024-01-17:
### Discussed in https://github.com/raspiblitz/raspiblitz/discussions/4355

<div type='discussions-op-text'>

<sup>Originally posted by **Nerfzooka** January  7, 2024</sup>
Before I show you a wall of debug text, take a look at what AI thinks a PCB centerpiece looks like:
[image]

Allright!  The old rpi node was running slow so I decided to give it a new sd card.  My wallets are set to connect to my node but they can't sync up.  The wallets say "no connection", the node says Electrum : Not running.  

I tried to add a "sudo journalctl -u electrs" output but it wouldn't fit in the discussion window.

Here's a debug output:  
*************************************************************** * RASPIBLITZ DEBUG LOGS  *************************************************************** blitzversion: 1.10.0 chainnetwork: bitcoin / main  20:09:27 up 1 day,  2:38,  1 user,  load average: 0.99, 1.20, 1.19  *** FAILED SERVICES *** list any servcies with problems: sudo systemctl list-units --failed   UNIT             LOAD   ACTIVE SUB    DESCRIPTION ● bitcoind.service loaded failed failed Bitcoin daemon on mainnet  LOAD   = Reflects whether the unit definition was properly loaded. ACTIVE = The high-level unit activation state, i.e. generalization of SUB. SUB    = The low-level unit activation state, values depend on unit type. 1 loaded units listed.  *** SETUPPHASE / BOOTSTRAP *** see logs: cat /home/admin/raspiblitz.log setupPhase--> done state--> ready  *** BACKGROUNDSERVICE *** to monitor Background service call: sudo journalctl -f -u background  *** BLOCKCHAIN (MAINNET) SYSTEMD STATUS *** ● bitcoind.service - Bitcoin daemon on mainnet      Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)      Active: failed (Result: exit-code) since Sun 2024-01-07 17:21:00 GMT; 2h 48min ago     Process: 3536647 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)     Process: 3536666 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)         CPU: 222ms  Jan 07 17:21:00 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'. Jan 07 17:21:00 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.  *** LAST BLOCKCHAIN (MAINNET) ERROR LOGS *** sudo journalctl -u bitcoind -b --no-pager -n20 -- Journal begins at Thu 2023-09-21 22:59:15 BST, ends at Sun 2024-01-07 20:09:27 GMT. -- Jan 07 17:20:59 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE Jan 07 17:20:59 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'. Jan 07 17:20:59 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet. Jan 07 17:20:59 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 6. Jan 07 17:20:59 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet. Jan 07 17:20:59 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet... Jan 07 17:20:59 raspberrypi bitcoin.check.sh[3536651]: /bin/chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system Jan 07 17:20:59 raspberrypi bitcoin.check.sh[3536654]: sed: couldn't open temporary file /mnt/hdd/bitcoin/sedenDphp: Read-only file system Jan 07 17:20:59 raspberrypi bitcoin.check.sh[3536655]: sed: couldn't open temporary file /mnt/hdd/bitcoin/sedvl0I4r: Read-only file system Jan 07 17:20:59 raspberrypi bitcoin.check.sh[3536658]: sed: couldn't open temporary file /mnt/hdd/bitcoin/sedbRPC6E: Read-only file system Jan 07 17:20:59 raspberrypi bitcoind[3536666]: Error: Settings file could not be written: Jan 07 17:20:59 raspberrypi bitcoind[3536666]: - Error: Unable to open settings file /mnt/hdd/bitcoin/settings.json.tmp for writing Jan 07 17:20:59 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE Jan 07 17:20:59 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'. Jan 07 17:20:59 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet. Jan 07 17:21:00 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 7. Jan 07 17:21:00 raspberrypi systemd[1]: Stopped Bitcoin daemon on mainnet. Jan 07 17:21:00 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly. Jan 07 17:21:00 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'. Jan 07 17:21:00 raspberrypi systemd[1]: Failed to start Bitcoin daemon on mainnet.  *** LAST BLOCKCHAIN (MAINNET) INFO LOGS *** sudo tail -n 50 /mnt/hdd/bitcoin/debug.log 2024-01-07T16:08:47Z UpdateTip: new best=00000000000000000001c846ebb61f86e426b3eb89d700ffbe09330bb1b41582 height=824761 version=0x20600000 log2_work=94.651530 tx=948948620 date='2024-01-07T16:08:20Z' progress=1.000000 cache=307.1MiB(2150145txo) 2024-01-07T16:08:59Z Saw new header hash=00000000000000000000cd95c5aef11bfda45ab92b646eadbfe385cd4c08de1b height=824762 2024-01-07T16:08:59Z [net] Saw new cmpctblock header hash=00000000000000000000cd95c5aef11bfda45ab92b646eadbfe385cd4c08de1b peer=455 2024-01-07T16:09:01Z UpdateTip: new best=00000000000000000000cd95c5aef11bfda45ab92b646eadbfe385cd4c08de1b height=824762 version=0x3fff0000 log2_work=94.651544 tx=948951996 date='2024-01-07T16:08:49Z' progress=1.000000 cache=307.7MiB(2154869txo) 2024-01-07T16:10:09Z Saw new header hash=000000000000000000015f9a06caa0ee6cffc3fe8b654d9e2b4df6210bdab9a8 height=824763 2024-01-07T16:10:09Z [net] Saw new cmpctblock header hash=000000000000000000015f9a06caa0ee6cffc3fe8b654d9e2b4df6210bdab9a8 peer=455 2024-01-07T16:10:11Z UpdateTip: new best=000000000000000000015f9a06caa0ee6cffc3fe8b654d9e2b4df6210bdab9a8 height=824763 version=0x29a2e000 log2_work=94.651559 tx=948954811 date='2024-01-07T16:09:59Z' progress=1.000000 cache=308.3MiB(2158665txo) 2024-01-07T16:12:32Z New outbound peer connected: version: 70016, blocks=824763, peer=1766 (block-relay-only) 2024-01-07T16:20:18Z Saw new header hash=00000000000000000002124c17bbf75b5807bdf092cfbde335243bb8f2bb39a2 height=824764 2024-01-07T16:20:21Z UpdateTip: new best=00000000000000000002124c17bbf75b5807bdf092cfbde335243bb8f2bb39a2 height=824764 version=0x32868000 log2_work=94.651573 tx=948958511 date='2024-01-07T16:19:59Z' progress=1.000000 cache=309.5MiB(2167516txo) 2024-01-07T16:43:56Z New outbound peer connected: version: 70016, blocks=824764, peer=1801 (block-relay-only) 2024-01-07T16:45:41Z Saw new header hash=00000000000000000003860b9bb6e7705e01ff312e8483dc53fe24249464c6b7 height=824765 2024-01-07T16:45:41Z [net] Saw new cmpctblock header hash=00000000000000000003860b9bb6e7705e01ff312e8483dc53fe24249464c6b7 peer=455 2024-01-07T16:45:43Z UpdateTip: new best=00000000000000000003860b9bb6e7705e01ff312e8483dc53fe24249464c6b7 height=824765 version=0x225c8000 log2_work=94.651588 tx=948961739 date='2024-01-07T16:45:26Z' progress=1.000000 cache=311.6MiB(2182927txo) 2024-01-07T16:45:51Z New outbound peer connected: version: 70016, blocks=824765, peer=1804 (block-relay-only) 2024-01-07T16:46:18Z Saw new header hash=0000000000000000000267a992e2ccb9f30bf5f3e50a85e19f20cc635250f631 height=824766 2024-01-07T16:46:19Z New outbound peer connected: version: 70016, blocks=824765, peer=1806 (outbound-full-relay) 2024-01-07T16:46:20Z UpdateTip: new best=0000000000000000000267a992e2ccb9f30bf5f3e50a85e19f20cc635250f631 height=824766 version=0x2f036000 log2_work=94.651602 tx=948964846 date='2024-01-07T16:46:03Z' progress=1.000000 cache=312.1MiB(2186516txo) 2024-01-07T16:46:24Z New outbound peer connected: version: 70016, blocks=824766, peer=1807 (block-relay-only) 2024-01-07T16:48:51Z Socks5() connect to ***.onion:8333 failed: host unreachable 2024-01-07T16:51:00Z New outbound peer connected: version: 70016, blocks=824766, peer=1819 (outbound-full-relay) 2024-01-07T16:53:06Z Socks5() connect to ***.onion:8333 failed: general failure 2024-01-07T16:54:49Z New outbound peer connected: version: 70016, blocks=824766, peer=1824 (block-relay-only) 2024-01-07T16:59:19Z New outbound peer connected:

Thanks for looking, any ideas as to what's wrong?
</div>

Nerfzooka, 2024-01-17:
Here, the "sudo journalctl -u electrs" output boiled down, it's this message repeated about 780 times :

-- Boot 634e0ec47bcf4153b2e7fbcdd1d8140b --
Jan 14 16:43:40 raspberrypi systemd[1]: Started Electrs.
Jan 14 16:43:41 raspberrypi electrs[30991]: Starting electrs 0.10.0 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", ""), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, skip_block_download_wait: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.10.0 - the Electrum Rust Server on your RaspiBlitz", signet_magic: f9beb4d9, args: [] }
Jan 14 16:44:08 raspberrypi electrs[30991]: Error: electrs failed
Jan 14 16:44:08 raspberrypi electrs[30991]: Caused by:
Jan 14 16:44:08 raspberrypi electrs[30991]: 0: bitcoind RPC polling failed
Jan 14 16:44:08 raspberrypi electrs[30991]: 1: daemon not available
Jan 14 16:44:08 raspberrypi electrs[30991]: 2: JSON-RPC error: transport error: Couldn't connect to host: Resource temporarily unavailable (os error 11)
Jan 14 16:44:08 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Jan 14 16:44:08 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 14 16:44:08 raspberrypi systemd[1]: electrs.service: Consumed 10.939s CPU time.
Jan 14 16:45:09 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 1.
Jan 14 16:45:09 raspberrypi systemd[1]: Stopped Electrs.
Jan 14 16:45:09 raspberrypi systemd[1]: electrs.service: Consumed 10.939s CPU time.
Jan 14 16:45:09 raspberrypi systemd[1]: Started Electrs.
Jan 14 16:45:09 raspberrypi electrs[33467]: Starting electrs 0.10.0 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", ""), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, skip_block_download_wait: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.10.0 - the Electrum Rust Server on your RaspiBlitz", signet_magic: f9beb4d9, args: [] }
Jan 14 16:45:35 raspberrypi electrs[33467]: Error: electrs failed
Jan 14 16:45:35 raspberrypi electrs[33467]: Caused by:
Jan 14 16:45:35 raspberrypi electrs[33467]: 0: bitcoind RPC polling failed
Jan 14 16:45:35 raspberrypi electrs[33467]: 1: daemon not available
Jan 14 16:45:35 raspberrypi electrs[33467]: 2: JSON-RPC error: transport error: Couldn't connect to host: Resource temporarily unavailable (os error 11)
Jan 14 16:45:35 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Jan 14 16:45:35 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 14 16:45:35 raspberrypi systemd[1]: electrs.service: Consumed 9.434s CPU time.
Jan 14 16:46:35 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 2.
Jan 14 16:46:35 raspberrypi systemd[1]: Stopped Electrs.
Jan 14 16:46:35 raspberrypi systemd[1]: electrs.service: Consumed 9.434s CPU time.
Jan 14 16:46:35 raspberrypi systemd[1]: Started Electrs.
Jan 14 16:46:35 raspberrypi electrs[34901]: Starting electrs 0.10.0 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", ""), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 0.0.0.0:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, skip_block_download_wait: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs v0.10.0 - the Electrum Rust Server on your RaspiBlitz", signet_magic: f9beb4d9, args: [] }
Jan 14 16:47:02 raspberrypi electrs[34901]: Error: electrs failed
Jan 14 16:47:02 raspberrypi electrs[34901]: Caused by:
Jan 14 16:47:02 raspberrypi electrs[34901]: 0: bitcoind RPC polling failed
Jan 14 16:47:02 raspberrypi electrs[34901]: 1: daemon not available
Jan 14 16:47:02 raspberrypi electrs[34901]: 2: JSON-RPC error: transport error: Couldn't connect to host: Resource temporarily unavailable (os error 11)
Jan 14 16:47:02 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Jan 14 16:47:02 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 14 16:47:02 raspberrypi systemd[1]: electrs.service: Consumed 10.537s CPU time.
Jan 14 16:48:02 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 3.
Jan 14 16:48:02 raspberrypi systemd[1]: Stopped Electrs.
Jan 14 16:48:02 raspberrypi systemd[1]: electrs.service: Consumed 10.537s CPU time.

rootzoll, 2024-01-17:
So electrs seems to have problems calling `bitcoind` .. the bitcoin debug logs were hard to read so I pumped it thru ChatGPT and it says:

The log you provided gives some important insights into why `bitcoind`, the Bitcoin daemon, is not running on your Raspberry Pi. The key points to note are:

1. **Read-Only File System Errors**: 
   - `chgrp: changing group of '/mnt/hdd/bitcoin': Read-only file system`
   - `sed: couldn't open temporary file /mnt/hdd/bitcoin/sed...: Read-only file system`

   These lines suggest that the file system where the Bitcoin data is stored (`/mnt/hdd/bitcoin`) has become read-only. This is often a protective measure taken by the operating system when it detects some kind of file system corruption or I/O error, typically with the physical storage device (like an SD card or HDD/SSD).

2. **Bitcoin Daemon Unable to Write Settings File**:
   - `Error: Settings file could not be written:`
   - `Error: Unable to open settings file /mnt/hdd/bitcoin/settings.json.tmp for writing`

   This indicates that `bitcoind` is trying to write to the settings file but is unable to do so because of the read-only status of the file system.

3. **Repeated Failure and Restart Attempts**:
   The log shows that the `bitcoind` service has repeatedly tried and failed to start, indicated by the multiple entries of `Failed to start Bitcoin daemon on mainnet`.

4. **Other Blockchain Logs Appear Normal**:
   The logs related to blockchain updates and peer connections (`UpdateTip: new best=...`, `Saw new header hash=...`, `New outbound peer connected:...`) appear normal up to a certain point. This suggests that `bitcoind` was functioning correctly until it encountered the read-only file system issue.

Given these points, the primary issue seems to be with the file system or the storage device where your Bitcoin data is located. To address this, you can try the following steps:

- **Check and Repair the File System**: Use tools like `fsck` (for Linux file systems) to check and repair any errors on the file system. This should be done with caution and ideally after taking a backup, if possible.

- **Check the Health of the Storage Device**: If you're using an SD card or an HDD/SSD, check its health. SD cards, in particular, have a limited lifespan and can become unreliable over time.

- **Ensure Adequate Power Supply**: Insufficient power supply to Raspberry Pi and connected storage devices can cause file system corruption. Make sure your power supply is adequate and stable.

- **Backup and Restore**: If you have a backup of your Bitcoin data, you might need to restore it after fixing the file system or replacing the storage device.

- **Consult Raspberry Pi and Bitcoin Community Forums**: If the problem persists, consider seeking advice from Raspberry Pi and Bitcoin community forums where you might find others who have faced similar issues.

Remember that dealing with file system issues and potential data corruption should be approached with caution, especially when it involves valuable data like a Bitcoin blockchain.

closed 2024-01-18 by Nerfzooka