raspiblitz/raspiblitz #4377: [RPi5] Migrate Option from external SSD to NVMe base

rootzoll, 2024-01-17:
On recover with a fresh sd card if a Pi5 and a NVMe and an external SSD with the name "BLOCKCHAIN" it could offer an automated data migration.

This could be used to migrate from an old external SSD used on a RPi4 or legacy RPi5 setup to an Pironi style NVMe base. 

Or when moving from an 1TB NVMe to a 2TB NVMe later by connecting the old NVMe thru a cheap USB adapter like this one:
[image]

rootzoll, 2024-01-17:
Can somebody test if on the RaspberryPi5 with the Pimoroni NVMe Base an additional connected USB SSD (the one we used before or such an USB NVMe adapter) can still copy lots of dummy data correctly without drops in power?

neversleep404, 2024-01-17:
I am in the fortunate position of owning two Pi 5s, allowing me to use one for testing purposes. I will receive a Pineberry Pi HatDrive Top tomorrow. A few weeks later, I'm expecting the Pimoroni NVMe Base. I believe their performance will be comparable.

My initial step involves creating a copy of my current productive USB SSD. Following that, on the test device, I plan to install the latest 1.11.pre dev version on the NVMe. Then, using an SD card loaded with the most recent Pi OS, I'll proceed to copy the blockchain onto the NVMe partition. f the synchronization has already started, I delete the blocks that have already been synchronized.

I intend to replicate the blockchain multiple times, utilizing various tools and settings such as cp, rsync, and dd, to compare their outcomes. My anticipation is that cp will offer the fastest performance, rsync the most reliability, and dd the slowest speed.

I am very much open to suggestions regarding this setup.

neversleep404, 2024-01-18:
First off, copying the productive USB SSD onto the test USB SSD with rsync has already worked without any problems. Following are the results:
sent 732,594,025,718 bytes  received 721,296 bytes  295,103,624.17 bytes/sec
total size is 732,412,654,771  speedup is 1.00

real	41m21.883s
user	0m1.202s
sys	0m2.873s

neversleep404, 2024-01-31:
[image]

Copying the blockchain over USB or LAN is unfortunately not possible without temporary cooling eg. using a fan.

neversleep404, 2024-01-31:
After making some adjustments, chatGPT suggests the following:

```bash
#!/bin/bash

# NVMe Temperature Monitoring and Blockchain Copy Script for Raspberry Pi
# This script copies the Bitcoin blockchain data from one drive to another,
# while monitoring the temperature of the NVMe SSD. The source files will
# not be deleted after copying.

# Set the source and destination directories
SOURCE="/path/to/source/blockchain" # Replace with the path to the source directory
DEST="/path/to/destination/blockchain" # Replace with the path to the destination directory

# Set the NVMe device. Example: /dev/nvme0n1
NVME_DEVICE="/dev/nvme0n1" # Replace with your NVMe device

# Set the maximum allowed temperature (in Celsius)
MAX_TEMP=50

# Create a temporary file to keep track of copied files
COPIED_FILES="/tmp/copied_files.txt"
> "$COPIED_FILES"

# Main loop
while true; do
    # Read the temperature of the NVMe SSD
    TEMP=$(sudo smartctl -A $NVME_DEVICE | grep -i "Temperature:" | awk '{print $10}')

    # Check if the temperature is within the acceptable range
    if [ "$TEMP" -le "$MAX_TEMP" ]; then
        # Find new files to copy, excluding those already copied
        find $SOURCE -type f | grep -vFf "$COPIED_FILES" | head -n 100 > "/tmp/to_copy.txt"
        
        # Check if there are new files to copy
        if [ ! -s "/tmp/to_copy.txt" ]; then
            echo "Copying complete"
            break
        fi

        # Copy the files
        rsync -av --progress --files-from="/tmp/to_copy.txt" "$SOURCE" "$DEST"
        
        # Add copied files to the list of copied files
        cat "/tmp/to_copy.txt" >> "$COPIED_FILES"
    else
        # Pause if the temperature is too high
        echo "Pausing due to high temperature ($TEMP°C)"
        sleep 60 # Wait for 60 seconds
    fi
done

# Cleanup
rm "/tmp/to_copy.txt" "$COPIED_FILES"

# End of script

```

neversleep404, 2024-02-02:
Looks fine :-)

[image]

neversleep404, 2024-02-02:
Overheating is really an issue. Here's an optimized script. I recommend not sorting the files and also consider using an external fan for cooling the NVMe SSD.

```bash
#!/bin/bash

# NVMe Temperature Monitoring and Blockchain Copy Script for Raspberry Pi
# This script copies the Bitcoin blockchain data from one drive to another,
# while monitoring the temperature of the NVMe SSD. The source files will
# not be deleted after copying.

# Set the source and destination directories
SOURCE="/mnt/source/bitcoin/blocks" # Replace with the path to the source directory
DEST="/mnt/target/test"             # Replace with the path to the destination directory

# Set the NVMe device. Example: /dev/nvme0n1
NVME_DEVICE="/dev/nvme0n1"          # Replace with your NVMe device

# Set the maximum allowed temperature (in Celsius)
MAX_TEMP=50

# Temporary files
TO_COPY_FILE="/tmp/to_copy.txt"
COPIED_FILES="/tmp/copied_files.txt"

# Delete temporary files if they exist
[ -f "$TO_COPY_FILE" ] && rm "$TO_COPY_FILE"
[ -f "$COPIED_FILES" ] && rm "$COPIED_FILES"

# Create a new empty file for tracking copied files
> "$COPIED_FILES"

# Main loop
while true; do
    # Read the temperature of the NVMe SSD
    TEMP=$(sudo smartctl -A $NVME_DEVICE | grep "Temperature:" | awk '{print $2}')

    # Check if the temperature is within the acceptable range
    if [ "$TEMP" -le "$MAX_TEMP" ]; then
        # Find new files to copy, excluding those already copied
		cd $SOURCE
        find . -type f | sed 's|^\./||'| grep -xvFf "$COPIED_FILES" | head -n 50 > "$TO_COPY_FILE"
                
        # Check if there are new files to copy
        if [ ! -s "$TO_COPY_FILE" ]; then
            echo "Copying complete"
            break
        fi

        # Copy the files with absolute paths
        rsync -av --progress --files-from="$TO_COPY_FILE" "$SOURCE/" "$DEST/"

        
        # Add copied files to the list of copied files
        cat "$TO_COPY_FILE" >> "$COPIED_FILES"
    else
        # Count the remaining files
        cd $SOURCE
        REMAINING_FILES=$(find . -type f | sed 's|^\./||'| grep -xvFf "$COPIED_FILES" | wc -l)

        # Pause if the temperature is too high
        echo "Pausing due to high temperature ($TEMP°C). $REMAINING_FILES files remaining to copy."
        sleep 60 # Wait for 60 seconds
    fi
done

# Cleanup
rm "$TO_COPY_FILE" "$COPIED_FILES"

# End of script
```

neversleep404, 2024-02-02:
[image]

neversleep404, 2024-02-02:
[image]

neversleep404, 2024-02-02:
[image]

TrevorPhilipsBR, 2024-02-03:
What do you think about displaying the NVME temperature alongside the CPU temperature on the dashboard?

neversleep404, 2024-02-04:
> What do you think about displaying the NVME temperature alongside the CPU temperature on the dashboard?
The heat problem should only occur when transferring large amounts of data, eg. the blockchain. In later operation, the NVMe hardly has anything to do.

rootzoll, 2025-05-17:
OK with v1.12.0 durinhg SSH menu setup there is now the "FROMHHD" Migration option. It also allows for example the migration on RaspberryPi4 from a old 1TB SSD to a new 2TB SSD .. its just important that the new drive is bigger and that on a RaspberryPi4 the old SSD has additional power supply. More details to come.