raspiblitz/raspiblitz #4392: TLS access on bitcoinRPC via nginx

rootzoll, 2024-01-31:
see https://github.com/raspiblitz/raspiblitz/discussions/4380#discussioncomment-8325856

rootzoll, 2024-01-31:
ChatGPT proposed the following config:

```
# Load the necessary NGINX modules
load_module modules/ngx_stream_module.so;

events {}

stream {
    # SSL configuration
    server {
        listen 50002 ssl; # Port for NGINX to listen on with SSL
        ssl_certificate /path/to/your/certificate.pem; # Path to your SSL certificate
        ssl_certificate_key /path/to/your/privatekey.pem; # Path to your SSL certificate key
        ssl_protocols TLSv1.2 TLSv1.3; # Enable TLS v1.2 and v1.3

        # Forward the traffic to Bitcoin Core RPC server
        proxy_pass 127.0.0.1:8332; # Local Bitcoin Core RPC server address

        # Security enhancements
        ssl_prefer_server_ciphers on;
        ssl_ciphers 'ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES256-GCM-SHA384:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-CHACHA20-POLY1305:ECDHE-RSA-CHACHA20-POLY1305:DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384';
        ssl_session_cache shared:SSL:10m;
        ssl_session_timeout 10m;
    }
}
```

In this configuration:

    Replace /path/to/your/certificate.pem and /path/to/your/privatekey.pem with the actual paths to your SSL certificate and key files.
    The NGINX server listens on port 50002 for incoming SSL connections.
    Traffic is then proxied to the local Bitcoin Core RPC server at 127.0.0.1:8332.
    It’s set to use TLS 1.2 and 1.3 for secure communication.
    The ssl_ciphers directive is configured for strong encryption. You may need to adjust this depending on your specific requirements and the capabilities of the client systems that will connect.

To use this configuration:

    Install NGINX and ensure it has the ngx_stream_module (this is usually included by default in most NGINX installations).
    Place this configuration in the appropriate directory for NGINX configurations, often /etc/nginx/nginx.conf or a specific file in /etc/nginx/conf.d/.
    Restart NGINX to apply the changes.

openoms, 2024-02-02:
What would this be used for?

rootzoll, 2024-02-09:
It might be a not often requested way of communicate with bitcoind - but in the end seems just a missing configuration of the tools we already have part as the std package (nginx). To maximize use of a bitcoin-only node I think its good to provide.