raspiblitz/raspiblitz #4393: local build of `dev` (@81ebfb7) fails on mempool-frontend: "npm run build did not run correctly" -

david-bakin-sl, 2024-02-01:
**Describe the bug**

Trying to build `dev` locally on a Pi5 (4GB) (in order to get access to the nVME capability that 1.10 doesn't have) and have the following log tail:

```
# npm install for mempool explorer (frontend)
npm WARN deprecated popper.js@1.16.1: You can find the new Popper v2 at @popperjs/core, this package is dedicated to the legacy v1
npm WARN deprecated @npmcli/move-file@2.0.1: This functionality has been moved to @npmcli/fs
npm WARN deprecated querystring@0.2.0: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.

added 1483 packages, and audited 1484 packages in 1m

189 packages are looking for funding
  run `npm fund` for details

26 vulnerabilities (16 moderate, 7 high, 3 critical)

To address issues that do not require attention, run:
  npm audit fix

To address all issues possible (including breaking changes), run:
  npm audit fix --force

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
npm notice
npm notice New major version of npm available! 9.8.0 -> 10.4.0
npm notice Changelog: https://github.com/npm/cli/releases/tag/v10.4.0
npm notice Run npm install -g npm@10.4.0 to update!
npm notice

> mempool-frontend@2.5.0 build
> npm run generate-config && npm run ng -- build --configuration production --localize && npm run sync-assets && npm run build-mempool.js

> mempool-frontend@2.5.0 generate-config
> node generate-config.js

mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.5.0
mempool revision 57eddac7f
src/resources/config.js file not found, creating new config file
CONFIG:  (function (window) {
  window.__env = window.__env || {};
    window.__env.GIT_COMMIT_HASH = '57eddac7f';
    window.__env.PACKAGE_JSON_VERSION = '2.5.0';
  }(this));
src/resources/config.js file saved

> mempool-frontend@2.5.0 ng
> ./node_modules/@angular/cli/bin/ng.js build --configuration production --localize

✔ Browser application bundle generation complete.
                                 Killed
FAIL - npm run build did not run correctly, aborting (1)
```

**To Reproduce**

`sudo ./build_sdcard.sh` having changed that script so that

```
defaultBranch="dev"
```

Built on the Pi5 itself (running off the nVME drive).  OS is vanilla `Raspberry Pi reference 2023-12-05`.

david-bakin-sl, 2024-02-01:
Rerunning the script now leads to what seems to be early termination, as shown by the following log:

```

Reading package lists...
Building dependency tree...
Reading state information...
qrencode is already the newest version (4.1.1-1).
secure-delete is already the newest version (3.1-8).
fbi is already the newest version (2.10-4+b1).
msmtp is already the newest version (1.8.23-1).
unclutter is already the newest version (8-25).
xterm is already the newest version (379-1).
python3-pyqt5 is already the newest version (5.15.9+dfsg-1).
xfonts-terminus is already the newest version (4.48-3.1).
apache2-utils is already the newest version (2.4.57-2).
nginx is already the newest version (1.22.1-9).
python3-jinja2 is already the newest version (3.1.2-1).
socat is already the newest version (1.7.4.4-2).
libatlas-base-dev is already the newest version (3.10.3-13).
hexyl is already the newest version (0.8.0-2+b5).
autossh is already the newest version (1.4g-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
* Adding LND ...
# *** INSTALL LND 0.17.3-beta BINARY ***
# only binary install to system
# no configuration, no systemd service
lnd binary already installed - done
```

rootzoll, 2024-02-09:
You can find the v1.11.0RC2 here to download https://github.com/raspiblitz/raspiblitz/actions/runs/7732163172 under "Artifacts" download, unzip and then flash the image to sd card - happy testing :)

david-bakin, 2024-02-11:
Correct me if I'm wrong, but this image (v1.11.0RC2) expects to be run from the SD card, not the NVMe drive?  And in fact it destroys itself it you install it to the NVMe drive and boot from it there because the first thing it tries to do is reformat the NVMe drive, right!?

(not a problem if that's the way it works ... and I suppose it makes future upgrades easier too ... just not what I expected ...)

Also - does it `fsck` the root drive on every boot or is that something I somehow picked up somewhere else?

rootzoll, 2024-02-11:
Running raspiblitz from the NVMe is not a standard setup yet - its something we target for a later release: #4324

david-bakin, 2024-02-12:
And now I'm a bit stuck.  The NVMe drive works fine under Raspberry Pi OS BTW.  But when I boot 1.11.0rc2 I see the usual spew on my HDMI monitor, then the LCD shows the Raspiblitz splash screen, then the LCD moves to the TUI and the top line reads `Raspiblitz 1.11.04c2 / standby /` and there's a text box reading `Starting Raspiblitz / Please wait ...` where it gets stuck (an hour at least).  And it doesn't show up on my router as a connected device.

Don't know how to troubleshoot this.  Obviously it hasn't gotten far enough that I can `ssh` into it.