raspiblitz/raspiblitz #4398: Enhance the SYSTEM menu

openoms, 2024-02-02:
Thinking that the `debug` and `debug -l` shortcuts should be moved to the SYSTEM menu from REPAIR to be available to view and share the debug info.

Should check if the SSD health check from https://github.com/raspiblitz/raspiblitz/issues/2599 does work with the NVME drives and give it an option to run manually:
```
sudo smartctl -a -d sat /dev/sda
```

`btop` has become available in Debian 12 so could be added in a new option for   nice monitoring dashboard covering the cPU load, processes, memory, disk usage and network all in one:
[image]

rootzoll, 2024-02-14:
added for v1.11.0 release

rootzoll, 2024-02-14:
TODO for v1.11.1 >> discuss/move DEBUG-REPORT shortcuts (why nit both?)