raspiblitz/raspiblitz #4399: RPi5 SSD tests

openoms, 2024-02-03:
A hot example with the Suptronics / Geekworm X1002 bottom board:
[image]
[image]

neversleep404, 2024-02-04:
Indeed a hot Fellow. The heat should only occur when transferring large amounts of data. In later operation, the NVMe hardly has anything to do."
Try this script: https://github.com/raspiblitz/raspiblitz/issues/4377#issuecomment-1924031599
and a copper heat sink.
[image]
And I recommend a fan to speed up the copying process."

neversleep404, 2024-02-04:
I would buy this 
Acidalie M.2 SSD heatsink with Double Layer Aluminum and 4 Copper Heat Pipes Cooler for M.2 SSD 2280
So cool

openoms, 2024-02-14:
Possibly problematic enclosure: ORICO-AM2C3 Enclosure M.2

rootzoll, 2024-02-16:
To better monitor the ssd/nvme heat I added the smartctl temp info to LCD & debug report - see #4422

closed 2025-09-27 by rootzoll