raspiblitz/raspiblitz #4402: LNbits 0.12.0

rootzoll, 2024-02-08:
a maybe upgrade for next raspiblitz release 
https://www.nobsbitcoin.com/lnbits-v0-12-0/

rd2madnss, 2024-03-14:
Current Version is now 0.12.2

dennisreimann, 2024-03-23:
fyi: I upgraded my LNbits instance via `config.scripts/bonus.lnbits.sh install lnbits 0.12.3`. Afterwards the superuser can be set with `sudo vim /home/lnbits/lnbits/.env` and adding the line `SUPER_USER=seeTheAccountMenuToLookupYourUserId`.

One more `sudo systemctl restart lnbits` and it seems to work like a charm.

therealsupermax, 2024-04-07:
This is great. I'll give this a test @dennisreimann

therealsupermax, 2024-04-07:
> fyi: I upgraded my LNbits instance via `config.scripts/bonus.lnbits.sh install lnbits 0.12.3`. Afterwards the superuser can be set with `sudo vim /home/lnbits/lnbits/.env` and adding the line [...]

I just ran command 'config.scripts/bonus.lnbits.sh install lnbits 0.12.3' and said it was already installed?
Also how do I edit the .env file when running 'sudo vim /home/lnbits/lnbits/.env'?

Thanks. Getting more proficient in CLI.

rootzoll, 2024-06-02:
PR merged #4596 - RaspiBlitz v1.11.1 will have lnbits v0.12.8

closed 2024-06-02 by rootzoll