raspiblitz/raspiblitz #4406: Github public signing key expired

semaf, 2024-02-10:
Hey,

The public key of Github has expired on 16.01.2024 and not renewed.

pub   rsa2048/4AEE18F83AFDEB23 2017-08-16 [SC] [expired: 2024-01-16]
      5DE3E0509C47EA3CF04A42D34AEE18F83AFDEB23
uid                 [ expired] GitHub (web-flow commit signing) <noreply@github.com>

pub   rsa4096/B5690EEEBB952194 2024-01-16 [SC]
      968479A1AFF927E37D1A566BB5690EEEBB952194
uid                 [ unknown] GitHub <noreply@github.com>

https://github.com/web-flow.gpg -> 4AEE18F83AFDEB23 
https://keyserver.ubuntu.com/pks/lookup?search=968479A1AFF927E37D1A566BB5690EEEBB952194&fingerprint=on&op=index

Discussion: https://github.com/orgs/community/discussions/89880

rootzoll, 2024-02-10:
We addressed the problem with the expired Webflow key already for the v1.110.0 release.

semaf, 2024-02-11:
> We addressed the problem with the expired Webflow key already for the upcmming v1.11.0 release.

Can you point to the file where the fix is?

rootzoll, 2024-02-11:
Basically update the following script: https://github.com/raspiblitz/raspiblitz/blob/dev/home.admin/config.scripts/blitz.git-verify.sh

rootzoll, 2024-03-19:
Looks good now. Closing issue.

closed 2024-03-19 by rootzoll