raspiblitz/raspiblitz #4415: Bitcoin & Lightning Progress 100%, but Node still syncing. Unable to access funds.

gardenofedon, 2024-02-12:
**Describe the bug**
I had a node that has been offline for years. I have recently begun starting to recover my funds from the node. Initially, I was using an old version of Raspiblitz and was unable to get the node working. After reading through threads on here, I decided to do a fresh-install, use the lnd-rescue file to rebuild my node (but with the blockchain copied from my laptop). 

I wiped all raspiblitz drivers, re-etched the SD card, and dragged the bitcoin folder from my computer to the Raspiblitz ugreen hard drive. I followed all steps, and have run into the same issue I was experiencing before: Bitcoin Progress 100% (and connected to fluctuating numbers of peers, and Lightning Progress 100%... but "Node is syncing" still displays. I can connect to the node via RTL, but Under Chain - Bitcoin Mainnet it says "Not Synced to Chain" when I drag my mouse over the red circle. I have tried to force close channels via RTL, but even though it says "Close Channel Request has been submitted" the channels never move into "Pending." 

I just want to remove all funds from the node. This has been really frustrating. I would really appreciate any assistance on how best to get the node working so that I can force close all channels and recover my funds. Thank you.

**Debug file**

Generating debug logs. Be patient, this should take maximum 2 minutes ..
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 23:25:58 up 25 min,  3 users,  load average: 4.98, 4.91, 3.87

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-02-12 23:02:44 GMT; 23min ago
    Process: 6391 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6415 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 6500 (bitcoind)
      Tasks: 18 (limit: 9352)
        CPU: 16min 30.432s
     CGroup: /system.slice/bitcoind.service
             └─6500 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Feb 12 23:01:48 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Feb 12 23:02:44 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Wed 2023-05-03 04:23:19 BST, ends at Mon 2024-02-12 23:25:58 GMT. --
Feb 12 23:01:48 raspberrypi systemd[1]: Starting Bitcoin daemon on mainnet...
Feb 12 23:02:44 raspberrypi systemd[1]: Started Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-02-12T23:02:44Z loadblk thread start
2024-02-12T23:02:44Z torcontrol thread start
2024-02-12T23:02:44Z Bound to *.*.*.*:8333
2024-02-12T23:02:44Z Bound to *.*.*.*:8334
2024-02-12T23:02:44Z Loaded 2 addresses from "anchors.dat"
2024-02-12T23:02:44Z [tor] Got service ID hyhtm3cy7oac7b4tssg5zk32xq67xqrzd2pwykcetwxpayy7o6dzg6ad, advertising service ***.onion:8333
2024-02-12T23:02:44Z 2 block-relay-only anchors will be tried for connections.
2024-02-12T23:02:44Z init message: Starting network threads…
2024-02-12T23:02:44Z DNS seeding disabled
2024-02-12T23:02:44Z AddLocal(***.onion:8333,4)
2024-02-12T23:02:44Z net thread start
2024-02-12T23:02:44Z addcon thread start
2024-02-12T23:02:44Z opencon thread start
2024-02-12T23:02:44Z msghand thread start
[... 363 lines cut ...]
btc_default_port="8333"
btc_default_sync_percentage="100.00"
btc_default_error_short=""
btc_default_error_full=""
btc_default_sync_initialblockdownload="0"
btc_default_blocks_verified="830205"
btc_default_peers="13"
btc_default_activated="1"
btc_default_mempool_transactions="5559"
btc_default_online="1"
btc_default_running="1"
btc_default_ready="1"
btc_default_blocks_headers="830205"
btc_default_version="v25.0.0"
btc_default_synced="1"
*** POSSIBLE ERROR REPORTS ***

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.

amcmillion, 2024-05-11:
Encountered the same issue on v1.10 with a fresh OS install. I successfully funded a wallet & connected with a handful of peers and but got stuck in this 100% sync state roughly 4 days ago after a brownout caused a hard shutdown.

[image]

## Repairs attempted

### `REINDEX-UTXO` 

Causes an error that locked the device for a couple minutes before rebooting into the Raspiblitz GUI:

`Type or paste the decryption passwordC` -> enter valid password

```
Job for lightningd.service failed because the control process exited with error code.
See "systemctl status lightningd.service" and "journalctl -xeu lightningd.service" for details.
# waiting to unlock wallet (0) ...
# waiting to unlock wallet (5) ...
# waiting to unlock wallet (10) ...
```

It returned to the 100% Blockchain + 100% Lightning network sync state and didn't budge for 12 hours.

### `REINDEX-FULL`

I'm now attempting a full reindex and approaching 22% blockchain progress. This is less than ideal as I'd prefer to `SYNC` from my desktop node but I can't figure out how to access this from the `repair` menu.

## Next Steps

I have both an `lnd-rescue` and `cl-rescue` files but I don't have my on-chain wallet seed phrase. My plan was to do another fresh start (this was my 3rd or 4th attempt as I could never get v1.11 working) and I didn't bother writing down my seed phrase.

### Pulling funds from on-chain wallet while the blockchain is still syncing

This stuck state has me unable to pull my funds from my on-chain wallet. Attempting to get any info from it fails in this state:

```
admin@192.168.12.198:~ ₿ bitcoin-wallet info
Failed to load database path '/home/admin/.bitcoin'. Data is not in recognized format.
admin@192.168.12.198:~ ₿ bitcoin-wallet dump
Failed to load database path '/home/admin/.bitcoin'. Data is not in recognized format.
```

The FAQs make it sound like I'll be able to recover _both_ on-chain and Lightning states with the rescue approach:

```
If you don't have them anymore: go back to option "Recover LND data" (see above) and check all possible ways to recover data from the HDD
```

But without my seed phrase I would still have to wait for the blockchain to sync before I could sign a transaction. And I'd definitely be more comfortable emptying the wallet and starting from scratch since I can't access my seed phrase at this point, and with my slow internet it will likely be another two weeks before this fresh sync from remote peers will finish.

Any advice?

rootzoll, 2024-05-24:
Some remarks - that hopefully help:

- RaspiBlitz v1.11.0 is available ... please always use & try the latest version. 
- Funds are not handled by Bitcoin Core. If you funded the node your wallet is either with LND or CoreLightning. So bitcoin-wallet cannot help you here.
- If you lost the seeds make sure to keep this lnd/cl-rescuefiles very safe - it contains your wallet.

For further help I would need to know if you used LND or CoreLightning.

amcmillion, 2024-05-27:
Thanks @rootzoll . I'm running CoreLightning.

I've updated my Raspi4 to the official RaspiBlitz v1.11 fatpack build and it simply hangs at boot. I have no ability to SSH into the box at this point. This behavior is identical to my last attempt at running v1.11 on April 22, after which I rolled back to v1.10.

What I see roughly 24 hours after booting up on v1.11:

[image]

## Debug Logs
Here are my logs from my v1.10 build while it was stuck at the 100% sync state, prior to upgrading to v1.11

### termbin
The termbin log output looks oddly short: https://l.termbin.com/ahei

### Full debug log dump from terminal

```
Printing debug logs. Be patient, this should take maximum 2 minutes ..
grep: warning: * at start of expression
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.10.0
chainnetwork: bitcoin / main
 20:22:37 up 5 days,  8:49,  2 users,  load average: 2.19, 1.83, 1.67

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT             LOAD   ACTIVE SUB    DESCRIPTION
● fail2ban.service loaded failed failed Fail2Ban Service

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
1 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-05-19 11:35:28 PDT; 5 days ago
    Process: 6841 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 6868 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 6950 (bitcoind)
      Tasks: 18 (limit: 8737)
        CPU: 1d 16h 27min 29.918s
     CGroup: /system.slice/bitcoind.service
             └─6950 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Notice: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-05-25T00:56:49Z New outbound peer connected: version: 70016, blocks=844998, peer=3552 (block-relay-only)
2024-05-25T00:58:14Z New outbound peer connected: version: 70016, blocks=844998, peer=3553 (block-relay-only)
2024-05-25T01:05:12Z Saw new header hash=000000000000000000023af04ed97e990754dccac0772f9fd581840ace627a35 height=844999
[... 311 lines cut ...]
ln_default_wallet_onchain_balance=****
ln_default_online="0"
ln_default_activated="1"
ln_default_wallet_channels_pending="0"
btc_default_sync_initial_done="1"
btc_default_sync_initialblockdownload="0"
btc_default_error_short=""
btc_default_version="v25.0.0"
btc_default_sync_percentage="100.00"
btc_default_online="1"
btc_default_peers="16"
btc_default_blocks_headers="845014"
btc_default_activated="1"
btc_default_running="1"
btc_default_error_full=""
btc_default_mempool_transactions="85711"
btc_default_port="8333"
btc_default_ready="1"
btc_default_blocks_behind="0"
btc_default_blocks_data_kb="637091812"
btc_default_sync_progress="0.999999273433995"
btc_default_blocks_verified="845014"
btc_default_synced="1"
*** POSSIBLE ERROR REPORTS ***
```

keybackrecovery, 2026-11-10:
Hi @amcmillion,

I came across your RaspiBlitz recovery issue and was wondering if you ever managed to regain access to your funds.

If the problem is still unresolved, I'd be happy to see if I can help. We can continue privately by email if you prefer.

Best regards,

keybackrecovery, 2026-11-10:
Hi @gardenofedon,

I came across your RaspiBlitz issue about being unable to withdraw your funds from the Lightning node.

Did you ever manage to recover them, or are you still dealing with the same problem?

Just curious how things turned out.