raspiblitz/raspiblitz #4416: LND not starting, wallet not found after updating

bigsome, 2024-02-13:
Hi there,

I don't know if my node has gone... I had a power outage and the node got stuck with "5 restarts message". Then I updated from raspiblitz 1.9 to 1.10 and had some errors with bitcoind, lnd... so I flashed a new card. Right now I have no errors on bitcoind (it says is 100% synced) but LND is not starting, it says there's no wallet, channel.db size is 5Kb... multiple errors.

Error log:
https://l.termbin.com/q7tn

It looks bad... Any suggestions to recover the node? or my funds?

I have the rescue file prior to the first update, and another one from 1 year ago (this last year I didn't do anything with the node) the old one is like 70Mb, the last one 22Mb... Maybe is better to try a rescue from the old one, what do you think?

Thanks in advance!

openoms, 2024-02-14:
Don't rely on an old channel.db. Use your latest static channel backup (SCB). 
You can extract the one from the rescue file as well if you have overwritten the SDcard and did not save anywhere else.

bigsome, 2024-02-17:
Thanks! Finally got the node recovered... Just rebooting and a little bit of patience. Now seems to work even faster. Although, I think I'm going to start with a new SDD from scratch. I don't feel 100% comfortable with my hardware.

closed 2024-02-17 by bigsome