raspiblitz/raspiblitz #4418: LNDg - Can not login with RPC (Password B)

bvolution, 2024-02-13:
**Describe the bug**
I cant log into lndg (webbrowser ip.to.my.raspi:8888) with my RPC Password B.
In the past LNDg login worked just fine. I did work with my node for a few weeks, I am not
sure what could have changed.

I tried to uninstall and reinstall lndg (keeping my data base though).
I also tried to copy the RPC Password directly from the bitocin.conf 
`sudo nano /mnt/hdd/bitcoin/bitcoin.conf`

However the error remains.

[image]

I would highly appreciated any tips or suggestions

bvolution, 2024-02-13:
Update: In the past I have manually changed my Password B (RPC). However this does not seem to include to also change the password for LNDg. I was able to fix it via

`sudo ./config.scripts/bonus.lndg.sh setpassword <mynewpassword>`
Which should maybe be added to the menu -> Password -> Change Password

rootzoll, 2024-02-15:
OK - created a new branch 4418-lndg to prepare a fix. 
Password should now be changed this way - but still needs to be tested.

BUT I ran into install problems when testing for RaspiBlitz v1.11.0 release. @allyourbankarebelongtous please take a look here: https://github.com/raspiblitz/raspiblitz/blob/4418-lndg/home.admin/config.scripts/bonus.lndg.sh

First the LNDg version 1.7.0 failed with this when running after install on gunicorn logs:
[image]

Then I saw there is a new version of LNDg 1.8.0 lets use that one - but that hangs even during install when running the "jobs.py":
[image]

If this gets not resolved may consider deactivating LNDg for next RaspiBlitz coming soon. Would love not to. 

TODO: fix install of v1.8.0
TODO: test new password change

rootzoll, 2024-02-23:
@allyourbankarebelongtous can you help fixing this (see above) for upcoming RaspiBlitz release?

rootzoll, 2024-02-23:
@cryptosharks131 can you help fixing this (see above) for upcoming RaspiBlitz release?

cryptosharks131, 2024-02-25:
Hi @rootzoll the `jobs.py` looks good above but in v1.8.0 it will no longer exit after each run with the addition of the `controller.py` file. You can now run the single `controller.py` file to start start all services (jobs, rebalancer, htlc-stream, trading) but running each file individually is still okay as well.

allyourbankarebelongtous, 2024-03-07:
Sorry folks, missed all this until @cryptosharks131 brought it up to me. I'll get on it and see what I can do.

rootzoll, 2024-03-07:
@allyourbankarebelongtous we still have a testing week before final release - if you can get a repair PR in by then happy to have it not deactivated for v1.11.0 release.

dragsugg, 2024-03-17:
With running lndg the update to the newest rc4 stucked. I've tried to unlock with lncli, but this didn't work either. The log shows this:

Sat Mar 16 21:11:36 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 21:11:56 2024 : [Data] : Starting data execution...
Sat Mar 16 21:11:56 2024 : [Data] : Error processing background data: <_InactiveRpcError of RPC that terminated with:
 status = StatusCode.UNKNOWN
 details = "wallet locked, unlock it to enable full RPC access"
 debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2024-03-16T21:11:56.159093546+00:00", grpc_status:2, grpc_message:"wallet locked, unlock it to enable full RPC access"}"
>
Sat Mar 16 21:11:56 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 21:12:16 2024 : [Data] : Starting data execution...

After unlocking the log shows following in a loop:

Sat Mar 16 15:08:13 2024 : [Data] : Starting data execution...Sat Mar 16 15:08:20 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:08:40 2024 : [Data] : Starting data execution...Sat Mar 16 15:08:47 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:09:07 2024 : [Data] : Starting data execution...Sat Mar 16 15:09:17 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:09:37 2024 : [Data] : Starting data execution...Sat Mar 16 15:09:53 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:10:13 2024 : [Data] : Starting data execution...Sat Mar 16 15:10:21 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:10:41 2024 : [Data] : Starting data execution...Sat Mar 16 15:10:48 2024 : [Data] : Data execution completed...sleeping for 20 seconds
Sat Mar 16 15:11:08 2024 : [Data] : Starting data execution...Sat Mar 16 15:11:15 2024 : [Data] : Data execution completed...sleeping for 20 seconds

Don't mind the times, I didn't find the right logfile and took it from someone else who had the same error

rootzoll, 2024-03-18:
The final relase was delayed and there might be still a week to submit a PR to fix the LNDg. @allyourbankarebelongtous is there a change to get this done in time?

rootzoll, 2024-03-25:
@dragsugg ok I might have fixed it. Please go in raspiblitz v1.11RC4or5 to terminal and run the following 3 commands:
```
/home/admin/config.scripts/bonus.lndg.sh off
patch code
/home/admin/config.scripts/bonus.lndg.sh on
```

Let me know if that fixes the LNDg

dragsugg, 2024-03-26:
Great, that works! I"ve deinstalled lndg on rc3, kept the data, updated to rc5 without problems, run the 3 commands and lndg works again. 👍

rootzoll, 2024-03-26:
OK closing this issue for final release.

closed 2024-03-26 by rootzoll

bvolution, 2024-04-16:
Thanks to all contributors sorry I could not do more. Thanks for fixing the issue.