raspiblitz/raspiblitz #4424: Add a repair web service

cstenglein, 2024-02-17:
Currently if the API or some other fails, there is currently no way to recover.

As a developer I would like to call to simple service which can find / fix problems with the API (/webserver if possible) so there is a way for the webui to recover if an error occurrs.

rootzoll, 2024-02-19:
I would say it is exactly that Repair-Sreen #4419 .. with the future planned "Error-Analyse" which should also find problems related to the webui.

cstenglein, 2024-02-19:
And how would an error analysis work from the webui, if the api e.g. can't be started? That's why a simple second service could be beneficial IMO. Don't know if there is a better solution than that.

rootzoll, 2024-02-19:
The whole repair/error-analysis can be a seperated API - its very RaspiBlitz specific anyway.

fusion44, 2024-02-19:
The service can share the login functionality with the main API and still be separate. I don't see a big issue with it and I'd actually prefer that.