raspiblitz/raspiblitz #4425: **issue of zerotier network id entering**  (rpi 5 with RaspiBlitz v1.11.0rc2)

dondelion, 2024-02-18:
**issue of zerotier network id entering**
Hello I'm running rpi 5 with RaspiBlitz v1.11.0rc2
I have a problem entering Zerotier network ID in settings menu --> i could fill  my network ID in the box but it couldn't confirm my entry instead it created a new line  :(

I've been running 2 pi4 raspiblitz without this problem.
please help I think this could be just a UI bug <3

[image]
[image]

rootzoll, 2024-02-19:
I was not able to reproduce this. I was able to copy & paste the networkid into the field and press ENTER to start the install/connection. I treid on MacOS ... you are on Windows?

As fallback you can go to the terminal and call the script directly with the networkid as parameter:
SSHMENU > EXIT > `./config.scripts/bonus.zerotier.sh on [networkid]`

dondelion, 2024-02-19:
Yes , i ssh via Putty on Windows. Not sure why this happened. btw thank you so much for a manual enter

dondelion, 2024-02-20:
I can navigate and enter text via putty on any menu except this one. I'll try to flash my raspiblitz and see the result.

kravens, 2024-03-09:
I have the same issue, using either powershell or windows terminal.
Also on an iPhone using the Termius app it does not work.
When pressing Enter or Return the menu goes to the next line instead of triggering the <Ok> signal.

rootzoll, 2024-03-10:
For v1.11.1 try switching between whiptail & dialog

rootzoll, 2024-03-10:
@dondelion & @kravens openoms did a PR to address this problem that can be tested with upcoming v1.11.0rc4 (hopefully availavle within the next 2 days) - please test and report if this fixes it for you.

rootzoll, 2024-03-19:
RC4 is out for testing. Looks good on my machine. Closing issue for release. Please comment/reopen if this error shoes again after v1.11.0rc4.

kravens, 2024-03-21:
On my Pi5 the menu is still an issue, it skips straight to this screen from the ''installing zerotier'' message when pressing Enter once: 
[image]
But I don't see the menu where I'm prompted to enter the network ID.

rootzoll, 2024-03-21:
@kravens please patch again to code in v1.11 branch and let me know if it now works.

kravens, 2024-03-21:
[image]
🎉 works now!

rootzoll, 2024-03-22:
Will be part of final release.

PR #4464 (2024-03-10): switch to dialog inputbox for zerotier
fixes: #4425
PR #4495 (2024-03-21): fix zerotier dialogs
fixes #4425
closed 2024-03-22 by rootzoll