raspiblitz/raspiblitz #4428: Cannot login to webinterface - Proxmox install

overcuriousity, 2024-02-20:
**Describe the bug**
I am unable to login to the webinterface through the IP of my instance. The login page appears, but after entering the password it says "An error occurred: Not Found". 
I am running my instance inside a Proxmox VM. Everything else seems to work, but there might be an issue with the Raspiblitz API i couldn´t narrow down further. It seems to have trouble to retreive some hardware information, which should be handled gracefully i.m.o.

**To Reproduce**
- Provision Raspiblitz dev Branch on Debian 12 VM in Proxmox
- Install Webinterface
- Try to login
- Fail

**Debug file**
Debug Output (From SSH):
```
Generating debug logs. Be patient, this should take maximum 2 minutes ..
sudo: vcgencmd: Befehl nicht gefunden
Redacting ..
# redacting file: /var/cache/raspiblitz/debug.log

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.11.0rc2
chainnetwork: bitcoin / main
 15:22:10 up 13 min,  1 user,  load average: 4,66, 4,30, 2,70

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-02-20 15:09:31 CET; 12min ago
   Main PID: 6166 (bitcoind)
      Tasks: 24 (limit: 19144)
     Memory: 12.6G
        CPU: 7min 19.157s
     CGroup: /system.slice/bitcoind.service
             └─6166 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Feb 20 15:* BTCNode systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Feb 20 15:* BTCNode systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
Feb 20 15:* BTCNode systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Feb 20 15:* BTCNode systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-02-20T14:09:31Z txindex thread start
2024-02-20T14:09:31Z Syncing txindex with block chain from height 253664
2024-02-20T14:09:31Z Leaving InitialBlockDownload (latching to false)
2024-02-20T14:09:40Z New block-relay-only v1 peer connected: version: 70016, blocks=831267, peer=1
2024-02-20T14:09:44Z New block-relay-only v1 peer connected: version: 70016, blocks=831267, peer=2
2024-02-20T14:09:46Z Saw new header hash=0000000000000000000323a3b7304ddba3e9f1a6bd762416d1a302096d1ca346 height=831267
2024-02-20T14:09:48Z New outbound-full-relay v1 peer connected: version: 70016, blocks=831267, peer=3
2024-02-20T14:09:55Z [i2p] Persistent SAM session 5e88e709d3 created, my address=jygaosber6ef66gxuwpjohqocgcntfemyco2ouerl6ss5xuhupca.b32.i2p:0
2024-02-20T14:09:55Z AddLocal(jygaosber6ef66gxuwpjohqocgcntfemyco2ouerl6ss5xuhupca.b32.i2p:0,4)
2024-02-20T14:10:09Z UpdateTip: new best=0000000000000000000323a3b7304ddba3e9f1a6bd762416d1a302096d1ca346 height=831267 version=0x3e000004 log2_work=94.744255 tx=967573606 date='2024-02-20T14:05:58Z' progress=0.999999 cache=2.8MiB(21222txo)
2024-02-20T14:10:10Z Syncing txindex with block chain from height 253722
[... 401 lines cut ...]
ln_default_sync_progress="100.00"
ln_default_recovery_done="0"
ln_default_channels_total="0"
btc_default_blocks_behind="0"
btc_default_error_full=""
btc_default_blocks_headers="831268"
btc_default_ready="1"
btc_default_error_short=""
btc_default_sync_initialblockdownload="0"
btc_default_sync_initial_done="1"
btc_default_mempool_transactions="3629"
btc_default_blocks_verified="831268"
btc_default_activated="1"
btc_default_version="v26.0.0"
btc_default_synced="1"
btc_default_online="1"
btc_default_blocks_data_kb="610679196"
btc_default_port="8333"
btc_default_sync_progress="0.9999971750408632"
btc_default_running="1"
btc_default_sync_percentage="100.00"
btc_default_peers="13"
*** POSSIBLE ERROR REPORTS ***

```

rootzoll, 2024-09-12:
We made some optimizations for Proxmox and when I tested the WebUI on my Proxmox testinstance it works fine, but I can see still similar errors in my API deug - so i opened an issue on the API repo to address this issue: https://github.com/fusion44/blitz_api/issues/271

overcuriousity, 2024-09-12:
I moved to a custom archlinux implementation (roughly following raspibolt) in the meantime, which runs good on proxmox.
However, I would be happy to assist with testing and improvements for raspiblitz on proxmox, if my assistance would benefit the project.

closed 2024-09-12 by overcuriousity

rootzoll, 2024-09-24:
@overcuriousity we stick so far to debian as a base - v1.11.3rc1 is getting closer (two weeks) any feedback & testing on Proxmox welcome.