raspiblitz/raspiblitz #4431: Expired Certificate in Electrs

wijg1, 2024-02-22:
When I attempt to connect Sparrow Wallet to my raspiblitz, I get the following error:

Could not connect:

Certificate expired at Sun Dec 24 17:00:12 CST 2023

NotAfter: Sun Dec 24 17:00:12 CST 2023

I updated Electrs on my raspiblitz and everything is the latest version as of 2/20/2024 (Raspiblitz 1.10.0, Bitcoin 25.0, CND .16.4-Beta). Why would I have this error?

wijg1, 2024-02-24:
Thank you for the response.  I am not a developer, just a Bitcoin user trying to connect my Sparrow to my Bitcoin node.  How can I fix the issue now so that I can connect to my Node?  I have been searching around the intrenet and have not found the solution yet.  Thanks again.

alex27riva, 2024-08-13:
Hi everyone, I have the same issue.
How to renew the certificate?

openoms, 2024-08-13:
To renew the self signed certificate (in case not using LND), run:
```
/home/admin/config.scripts/internet.selfsignedcert.sh reset
```

If the LND certificate is used that should autorenew on restart.

rootzoll, 2024-08-15:
Check if for a future release always a self-signed cert can be used/created with a very far away end date.

rootzoll, 2024-09-26:
These are the certs electrum uses on its SSL port:

https://github.com/raspiblitz/raspiblitz/blob/f11e529ff6ae720cdccf28ddac38a24d853c538b/home.admin/config.scripts/bonus.electrs.sh#L415-L416

Those nginx cert can either be the LND cert or the the one created by `internet.selfsignedcert.sh`

https://github.com/raspiblitz/raspiblitz/blob/f11e529ff6ae720cdccf28ddac38a24d853c538b/home.admin/config.scripts/blitz.web.sh#L128-L129

Checking `internet.selfsignedcert.sh` we can see that it creates a self-signed cert that should outdate about 10years after creation:

https://github.com/raspiblitz/raspiblitz/blob/f11e529ff6ae720cdccf28ddac38a24d853c538b/home.admin/config.scripts/internet.selfsignedcert.sh#L58-L59

**So the problem seems when the LND cert is used - for TEAMDISCUSS .. should we just always use the `internet.selfsignedcert.sh` one?**

_SIDENOTE: 

> Self-signed certificates and internal certificate authorities, the restrictions are generally more lenient:
> - Chrome and Edge do not enforce the 398-day limit for certificates from internal certificate authorities[1].
> - Safari restricts certificate validity for internal certificate authorities to a maximum of 825 days if issued after July 1, 2019 --> https://www.gradenegger.eu/en/chrome-and-safari-limit-ssl-certificates-to-one-year-2020/

In general we should not make our self-signed cert longer valid than 825days is we want to raise the bar._

rootzoll, 2024-10-01:
get rid if lnd cert for next release and see further feedback

rootzoll, 2024-10-03:
OK this will come with v1.11.3 - closing. Please reopen if still happening after v1.11.3 release.

closed 2024-10-03 by rootzoll