raspiblitz/raspiblitz #4436: [API/WebUI] Electrs will not be shown as installed

rootzoll, 2024-02-24:
With v1.11.0rc3 (with fixed API) Electrs is installed (thru ssh menues previously) but does but still shows in WebUI as uninstalled.

This is the SSH menus:
[image]

This is the WebUI:
[image]

@fusion could this still be a side effect from the API?

rootzoll, 2024-02-24:
Sidenote: Electrs is still building the index ... maybe thats why its not shown as installed?

fusion44, 2024-02-25:
This is because every other script returns the installed status as `installed=1`, but electrs returns `serviceInstalled=1`. I'll add a special path for this. We should unify this for the next major release. Not sure why I didn't catch this during development. Doesn't seem to have been changed recently.

Electrs: https://github.com/raspiblitz/raspiblitz/blob/ec67c362f755ad968001d10d7815406cd8309a1e/home.admin/config.scripts/bonus.electrs.sh#L39
RTL: https://github.com/raspiblitz/raspiblitz/blob/ec67c362f755ad968001d10d7815406cd8309a1e/home.admin/config.scripts/bonus.rtl.sh#L71

rootzoll, 2024-02-25:
@fusion44 I will change script to "installed"

rootzoll, 2024-02-25:
BTW it was not unifyed because we didnt prepare the electrs script for fatpack/webui yet. I will go thru.

rootzoll, 2024-02-25:
Prepared a PR #4438 to fix the install script - testing at the moment.

fusion44, 2024-02-25:
Ok perfect. I'll adapt once the pr is merged.

I can't use the conf file, because I need the connection info too. Maybe we can cache this into redis in the future.

rootzoll, 2024-02-25:
OK the "Retry Later" strategy worked - electrs is now installing and the Electrs is now showing up in the Applications section of the webUI when openend

But I now see two "Open" Buttons on the EletcRS item - see:
[image]

@cstenglein I guess this is still to be fixed on the WebUI, correct?

rootzoll, 2024-03-19:
This looks good now. Closing issue.

closed 2024-03-19 by rootzoll