raspiblitz/raspiblitz #4437: api: http 500 error on list-all-tx

cstenglein, 2024-02-24:
```
Feb 24 20:14:42 raspberrypi python[2296698]: Task exception was never retrieved
Feb 24 20:14:42 raspberrypi python[2296698]: future: <Task finished name='Task-4051' coro=<warmup_new_connections() done, defined at /home/blitzapi/blitz_api/app/main.py:278> exception=AttributeError("'HTTPException' object has no attribute 'dict'")>
Feb 24 20:14:42 raspberrypi python[2296698]: Traceback (most recent call last):
Feb 24 20:14:42 raspberrypi python[2296698]:   File "/home/blitzapi/blitz_api/app/main.py", line 303, in warmup_new_connections
Feb 24 20:14:42 raspberrypi python[2296698]:     _send_sse_event(id, SSE.SYSTEM_INFO, res[0].dict()),
Feb 24 20:14:42 raspberrypi python[2296698]:                                          ^^^^^^^^^^^
Feb 24 20:14:42 raspberrypi python[2296698]: AttributeError: 'HTTPException' object has no attribute 'dict'
Feb 24 20:14:44 raspberrypi python[2296698]: 2024-02-24 20:14:44 | ❌ | service.py:210 | Initializing Bitcoin Core (loading, verifying blocks or starting network threads etc)
```

rootzoll, 2024-03-07:
This is also still a problem with the v1.11.0rc3

[image]

@fusion44 is there already an issue on the API repo addressing this or schould I open a different one?

fusion44, 2024-03-10:
> This seems the corresponding issue on the API repo: [fusion44/blitz_api#247](https://github.com/fusion44/blitz_api/issues/247)

They are two different problems. This issue should be fixed in the dev branch.

rootzoll, 2024-03-19:
This looks good now. Closing for release.

closed 2024-03-19 by rootzoll