raspiblitz/raspiblitz #4445: New Install with image from 2024-02-25 and CL Resultat in a not running CL

neversleep404, 2024-02-27:
Build raspiblitz-arm64-rpi-image-2024-02-25-45c5c0f
Fresh install of a new Pi 5 with SD-Card and NVMe results in an not running CL.
Shows "missing Data" in red.
On the LCD Display is also the green onion Link missing.

A Update from a running PI 5 from a build from last week okkures in following errors:

[image]

[image]

openoms, 2024-02-27:
Can you share the CLN error log please:
```
sudo journalctl -fu lightningd -n50
```

neversleep404, 2024-02-28:
This screenshots are from a fresh and clean setup of raspiblitz-arm64-rpi-image-2024-02-25-45c5c0f.
Blockchain existet and is used. All other data are deleted (in the setup process). In the setup I used CL without any backup.
These are the Results:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

openoms, 2024-02-28:
Found the cause< it is the backup plugin failing.
The temporary solution is to deactivate it with running:
```
config.scripts/cl-plugin.backup.sh off
```

neversleep404, 2024-02-28:
> Found the cause, it is the backup plugin failing. The temporary solution is to deactivate it with running:
> 
> ```
> config.scripts/cl-plugin.backup.sh off
> ```

Ok. It worked.
Shoud I try the raspiblitz-arm64-rpi-image-2024-02-28-ede510f with a fresh and clean installation?

neversleep404, 2024-02-28:
raspiblitz-arm64-rpi-image-2024-02-28-ede510f works fine

openoms, 2024-02-28:
> raspiblitz-arm64-rpi-image-2024-02-28-ede510f works fine

ok thanks for testing. Is the backup plugin on now?
```
cl plugin list
```
to be sure can also run:
```
config.scripts/cl-plugin.backup.sh on
```
to reinitialize the plugin.

rootzoll, 2024-03-19:
Closing for release.

PR #4446 (2024-02-29): cln backup plugin update
Attempting to fix #4445
closed 2024-03-19 by rootzoll