raspiblitz/raspiblitz #4449: Raspiblitz SSH Screen shows "features" when scanning blocks

cstenglein, 2024-03-01:
When LND is scanning blocks, the raspiblitz screen displays "features".

[image]

rootzoll, 2024-03-07:
This is really on the LCD screen ... because this comes from this place which is normally just when run in the SSH terminal:
https://github.com/raspiblitz/raspiblitz/blob/003230d1377333341d57568628a5d8024793cfa7/home.admin/00mainMenu.sh#L226-L227

rootzoll, 2024-04-16:
solved on startup but seen it sometimes on running still - pushing to v1.11.1

rootzoll, 2024-09-27:
prepared branch `4449-features` but need to prepare scenario for testing by keeping lnd stoped for several hours

rootzoll, 2024-09-28:
OK was able to reproduce and found the `features` string in:
```
admin@192.168.178.143:~ ₿ cache export | grep feature
ln_default_address="features"
ln_lnd_mainnet_address="features"
```

cstenglein, 2024-10-07:
Re-opening it since it was only partially fixed.

In the rc1 it now displays "null" instead of "features"

cstenglein, 2024-10-09:
[image]

rootzoll, 2025-05-18:
I think I found the "null" thing - closing for v1.12.0 release

PR #4762 (2024-10-02): #4449 Fix `features` bug
fixes #4449