raspiblitz/raspiblitz #4451: LNDg: CSRF verification failed.

robert-winkler, 2024-03-03:
I installed LNDg from the RaspiBlitz Menu v1.10.0 (upgraded from V.1.9.0) 

After installation, the login page is accessible, but after entering lndg-admin:password the following error occurs: 

~~~
Forbidden (403)
CSRF verification failed. Request aborted.
More information is available with DEBUG=True.
~~~

I tried after removing the /app-data directory on the HDD, and reinstalling, but without success. 

Any idea?

rootzoll, 2024-03-07:
see #4418 - LNDg will be deactoivated for repair in next release.

rootzoll, 2024-03-19:
Might put out a bounty for repair.

rootzoll, 2024-03-25:
Please wait for release v1.11.0 ... if LNDg still does not work on that version please write here.

rootzoll, 2024-03-26:
OK looks like we got lndg fixed as part of v1.11.0 - closing for final release.

closed 2024-03-26 by rootzoll