raspiblitz/raspiblitz #4452: LCD Rotate

dragsugg, 2024-03-03:
Hello, 

I just can't manage to rotate my LCD display. I have tried all the configurations in boot/config.txt, but none of them change anything. 

With the command in the settings under LCD rotate nothing happens either. 

I just want my display to be the right way up when the power cable is up on the Pi5. 

Maybe someone has an idea. I would be delighted! 😊

https://amzn.eu/d/1yjPIZe I use the Elegoo Display, which is recommended and worked with the Pi4 before

rootzoll, 2024-03-06:
Interesting - the old way we do it thru the config.txt seems not to work anymore. So i can confirm that bug. Found no quick fix for the upcoming release - so have to push it to v1.11.1 for later deeper research. Might take a while.

rootzoll, 2024-03-07:
@fusion44 if you want to revive the TouchscreenUI development (maybe as focus project) rotating would be also a requested feature here.

rootzoll, 2024-06-02:
Just tested again and I think we already fixed this v1.11.0  - rotating works on my test machine .. closing issue. If it will still not work with latest v1.11.1rc1 - please reopen.

closed 2024-06-02 by rootzoll