raspiblitz/raspiblitz #4456: Local Packer Build fails

rootzoll, 2024-03-07:
@openoms my local packer build fails  .. can you take a look:

```
# Download the packer-builder-arm plugin
Cloning into 'packer-builder-arm'...
remote: Enumerating objects: 2090, done.
remote: Counting objects: 100% (845/845), done.
remote: Compressing objects: 100% (287/287), done.
remote: Total 2090 (delta 668), reused 658 (delta 558), pack-reused 1245
Receiving objects: 100% (2090/2090), 14.18 MiB | 10.55 MiB/s, done.
Resolving deltas: 100% (1219/1219), done.
fatal: detected dubious ownership in repository at '/media/user/NTFS/raspiblitz/ci/arm64-rpi/packer-builder-arm'
To add an exception for this directory, call:

	git config --global --add safe.directory /media/user/NTFS/raspiblitz/ci/arm64-rpi/packer-builder-arm

# Build the packer-builder-arm plugin
error obtaining VCS status: exit status 128
	Use -buildvcs=false to disable VCS stamping.
make: *** [Makefile:85: arm64-rpi-lean-image] Error 1
```

This is waht the GPT gave me:
```
The error error obtaining VCS status: exit status 128 suggests that there's a problem related to Version Control System (VCS) stamping during the build process of the packer-builder-arm plugin. VCS stamping is often used to include version control information (like commit hash, branch name) in the build artifacts for traceability.

Solution:
As the error message hints, you can disable VCS stamping by adding -buildvcs=false to your build command. This can help bypass the issue if VCS stamping isn't crucial for your use case. If you're using a Makefile to automate the build (as suggested by the make: *** [Makefile:85: arm64-rpi-lean-image] Error 1 message), you might need to modify the relevant commands within your Makefile or pass this flag appropriately, depending on how the Makefile is structured.

Without the exact details of your build command or the Makefile content, I would suggest looking for the part of the Makefile that triggers the build process of the packer-builder-arm plugin and see if you can incorporate the -buildvcs=false flag there.
```

openoms, 2024-03-07:
Can you try to run the build as sudo (run `sudo su ` beforehand )?

Also what is this exactly? The make command?

rootzoll, 2024-03-07:
> Can you try to run the build as sudo (run `sudo su ` beforehand )?

Tried - did not changed the outcome - same error. 

> Also what is this exactly? 

Following the new build procedure documented here: https://github.com/raspiblitz/raspiblitz/blob/dev/FAQ.dev.md#steps-to-create-raspberrypi-images-with-packer

rootzoll, 2024-03-07:
The new local build script I use is doing the make here: https://github.com/raspiblitz/raspiblitz/blob/87d63e5f32003a41dc686f1ee68729848c8424cf/ci/packer.sh#L154

rootzoll, 2024-03-07:
the `packer.sh` script is deleting any old downloaded repo already and gets it fresh - see here:
https://github.com/raspiblitz/raspiblitz/blob/87d63e5f32003a41dc686f1ee68729848c8424cf/ci/packer.sh#L79

The "dubious ownership" warning is not my concern here (that is checked by the commit hash) it fail because of the "VCS status" .. but I can try to add the "-buildvcs=false" parameter to the make command tomorrow.

openoms, 2024-03-07:
You might be downloading and deleting the raspiblitz repo again inside the repo

This should work:
```
shasum -a 256  /media/user/NTFS/raspiblitz/ci/packer.sh
sudo rm -rf /media/user/NTFS/raspiblitz
curl -O -L https://raw.githubusercontent.com/raspiblitz/raspiblitz/dev/ci/packer.sh
shasum -a 256 packer.sh

sudo bash ./packer.sh dev arm min
```

rootzoll, 2024-03-08:
ok - seems to work now

PR #4460 (2024-03-08): Fix #4456 local packer build

fixes #4456
commit 2de42dc7 (2024-03-08): Fix #4456 local packer build (#4460)
commit 2f0abcf7 (2024-03-08): fix #4456 packer local build (#4461)