raspiblitz/raspiblitz #4457: Watchtower Plugin CLN v23.11.2 failure stops CLN

neversleep404, 2024-03-08:
Installing Watchtower on RC3 CLN v23.11.2 on a PI 5 stops with a failure. 
Switching the Plugin off and CLN works again "cl-plugin.watchtower-client.sh off mainnet".

[image]

[image]

[image]

[image]

openoms, 2024-03-08:
In
```
sudo journalctl -fu lightningd
```
getting the wrong config option:
```
raspiblitz-amd64 sh[2244002]: lightningd: Config file /home/bitcoin/.lightning/config line 16: watchtower-proxy=127.0.0.1:9050: unknown option
```

openoms, 2024-03-08:
Instead of `watchtower-proxy=127.0.0.1:9050` it needs to be `proxy=127.0.0.1:9050`
as in https://github.com/talaia-labs/rust-teos/tree/master/watchtower-plugin#core-lightning-cln-config

neversleep404, 2024-03-11:
raspiblitz-arm64-rpi-image-2024-03-10-b8fbb54 works fine on my Test System. Watchtower client plugin did not stops CLN.

rootzoll, 2024-03-19:
Closing for Release.

closed 2024-03-19 by rootzoll