raspiblitz/raspiblitz #4466: Check if Backup USB Stick makes trouble with HDD

rootzoll, 2024-03-11:
Recheck Setup

```
So I can report one finding.  I keep a stubby old thumbdrive in one of the two slower USB ports on the PI.  This is for scheduled backups.  But as long as that is plugged in it seems to make the system recognizing the USB 3 attached HDD fail more than succeed.  I have run the test a number of times with the same result.  With the thumb drive in no HDD shows up 80% or so.  Removed HDD mounts 100%.
```

cAP5L0CK, 2024-03-11:
I am  the original reporter.  I have
-RPi 5 8gb.  Stock. No mods or overclocking.
-1TB SATA SSD in a USB case (no external power) attached to a USB3 port
-65W Laptop style power supply
-16G Verbatum USB thumb drive

When the thumb drive is inserted in one of the 2.1 USB ports (the slower ones) then the HDD does not auto mount in most cases.  If it is not the HDD mounts every time.

cAP5L0CK, 2024-03-15:
Will be replacing the 65w supply with the official 27w 5.1V, 5A PD type to see if this solves the issue.

cAP5L0CK, 2024-03-15:
PSU replaced.  This was the source of the issue in my opinion (and I believe rootzoll also predicted this early on.

sudo dtc -I fs -f /proc/device-tree/chosen/power/  

USED TO PRODUCE:
max_current = <0xbb8 >;  (or 3000 which is 3A)

NOW it produces:
max_current = <0x1388>;  (or 5000 which is 5A)

Almost all power sullies, even those that are much more powerful do not produce 5A @ 5.1V but produce 3A @ 5.1V

The official adapter delivers this, and there are a handful of 3rd party ones available.  I bought this one:
https://www.amazon.com/dp/B0CT5B5ZLP

Incidentally this decreases the startup time of the PI to something MUCH quicker as well.  I assume when it is using the 3A supplies it can boot, but the 100% fan combined with the draw on boot can cause my issue as well as it not seeing the USB drive early enough in the bootcycle to continue.

Another note.  When using an underamped supply  If after the pi has started looking for the drive you disconnect/reconnect it it will find it immediately and boot normally.  But 4/5times it hangs without this trick.

Obviously getting the correct power for this little thing cannot be overstated.  It will run on 3A (like a RPi4 adapter, or my 65W laptop charger) but during times of stress r/w to the HDD will fail.

rootzoll, 2025-06-04:
Rechecked USB backup drive for upcoming RaspiBlitz v1.12.0 - PR #5044 ready to merge

PR #5044 (2025-06-04): #4466 adapt USB backup thumb drive 
closes #4466
closed 2025-06-04 by rootzoll