raspiblitz/raspiblitz #4468: WebUI/SSH sync "Setup Done"

rootzoll, 2024-03-12:
On testing rc4 with fatpack recovering thru WebUI ... WebUI is still spinning "Setup System" while SSH-LCD shows "Setup Done". Recheck if reproducable.

rootzoll, 2024-03-12:
Also WebUI is not showing up after reboot "403 Forbidden" & electRS seems not running

debug report
```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.11.0rc4
chainnetwork: bitcoin / main
 02:30:42 up 4 min,  2 users,  load average: 3.34, 2.21, 0.96

*** FAILED SERVICES ***
list any servcies with problems: sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-03-12 02:27:05 GMT; 3min 37s ago
    Process: 5345 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 5359 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
   Main PID: 5381 (bitcoind)
      Tasks: 20 (limit: 9440)
        CPU: 4min 22.227s
     CGroup: /system.slice/bitcoind.service
             └─5381 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin

Mar 12 02:26:52 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Mar 12 02:27:05 raspberrypi systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
Mar 12 02:26:52 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon on mainnet...
Mar 12 02:27:05 raspberrypi systemd[1]: Started bitcoind.service - Bitcoin daemon on mainnet.

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2024-03-12T02:27:03Z Verification progress: 33%
2024-03-12T02:27:03Z Verification progress: 50%
2024-03-12T02:27:03Z Verification progress: 66%
2024-03-12T02:27:04Z Verification progress: 83%
2024-03-12T02:27:04Z Verification progress: 99%
2024-03-12T02:27:04Z Verification: No coin database inconsistencies in last 6 blocks (14436 transactions)
2024-03-12T02:27:04Z  block index           11647ms
2024-03-12T02:27:04Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2024-03-12T02:27:04Z Opened LevelDB successfully
2024-03-12T02:27:04Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2024-03-12T02:27:04Z Setting NODE_NETWORK on non-prune mode
2024-03-12T02:27:04Z block tree size = 834301
2024-03-12T02:27:04Z nBestHeight = 834297
2024-03-12T02:27:04Z initload thread start
2024-03-12T02:27:04Z txindex thread start
2024-03-12T02:27:04Z Bound to *.*.*.*:8333
2024-03-12T02:27:04Z txindex is enabled at height 834297
2024-03-12T02:27:04Z txindex thread exit
2024-03-12T02:27:04Z Bound to *.*.*.*:8334
2024-03-12T02:27:04Z torcontrol thread start
2024-03-12T02:27:04Z Loaded 0 addresses from "anchors.dat"
2024-03-12T02:27:04Z 0 block-relay-only anchors will be tried for connections.
2024-03-12T02:27:04Z init message: Starting network threads…
2024-03-12T02:27:04Z DNS seeding disabled
[... 425 lines cut ...]
ln_default_fees_total="0"
ln_default_channels_pending="0"
ln_default_running="1"
ln_default_peers="7"
btc_default_synced="1"
btc_default_sync_initial_done="0"
btc_default_port="8333"
btc_default_blocks_headers="834297"
btc_default_blocks_behind="0"
btc_default_error_short=""
btc_default_blocks_data_kb="616587888"
btc_default_ready="1"
btc_default_mempool_transactions="30047"
btc_default_sync_initialblockdownload="0"
btc_default_sync_percentage="100.00"
btc_default_sync_progress="0.9999921320047336"
btc_default_activated="1"
btc_default_version="v26.0.0"
btc_default_error_full=""
btc_default_peers="1"
btc_default_running="1"
btc_default_online="1"
btc_default_blocks_verified="834297"
*** POSSIBLE ERROR REPORTS ***
```

rootzoll, 2024-03-12:
OK .. the node that was recovered had `blitzapi=off` and seemed to have `deactivated` the fatpack active WebUI. Will verify if that was the cause by now witching on.

rootzoll, 2024-03-18:
So this seems to be the palce where during fatpack recovery the WebUI & API gets deinstalled (as it should by node config):
https://github.com/raspiblitz/raspiblitz/blob/f1342ebcfd12b2e81e93e328c1e6393dd73c72be/home.admin/_provision_.sh#L199-L200

The idea is to send as last status/message at least the info that the WebUI gets deactivated so its for the user to see.
@cstenglein if this works - it may make sense for v1.11.1 to have a special screen in the SetupWebUI on status "DeactivateAPI/WebUI" that gives a bit more information.

cstenglein, 2024-03-18:
Provisioning Messages get forwarded and displayed automatically on the webui.
Or do you mean some extra setting?

rootzoll, 2024-03-18:
@cstenglein yes its works good enough for now - the webUI will stop after my modifications with:
[image]

And taking into account that its a bit of an edge case I think we covered it good enough with that.

rootzoll, 2024-03-18:
Did a bit nicer text now:
[image]

rootzoll, 2024-03-19:
OK closing issue for release.

PR #4486 (2024-03-18): Info about API/WebUI deactivation on fatpack provision 
fixes #4468
commit 24a3c06a (2024-03-18): Info about API/WebUI deactivation on fatpack provision  (#4486)

fixes #4468
closed 2024-03-19 by rootzoll