raspiblitz/raspiblitz #4483: 1.11.0 orc4 Zeroteir won't install

glensidecross, 2024-03-17:
After updating to 1.11.0 orc4 Zeroteir didn't work. 

I tried uninstall and reinstall Zeroteir. Upon reinstalling, the install hangs on: zeroteir setting changed ..

If I hit control c to return to command prompt, a dialogue box says 'Your Raspiblitz has joined the ZeroTier network. 

Zerotier still does not work, and if I check settings, Zerotier has no asterix indicating that it is not installed. 

I am using a Raspberry pi 5 8GB with external 1tb SSD

rootzoll, 2024-03-18:
I was not able to reproduce - my test with zerotier worked .. also with uninstall/reinstall.

Can you try to install from terminal: `/home/admin/config.scripts/bonus.zerotier.sh on [networkid]` and post the complete install log output here (just blackout your networkid in the logs).

glensidecross, 2024-03-18:
That worked. 

# *** INSTALL ZeroTier ***
gpg: key 1657198823E52A61: "ZeroTier, Inc. (ZeroTier Support and Release Signing Key) <contact@zerotier.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: Signature made Mon 30 Oct 2023 19:07:38 GMT
gpg:                using RSA key 74A5E9C458E1A431F1DA57A71657198823E52A61
gpg:                issuer "contact@zerotier.com"
gpg: Good signature from "ZeroTier, Inc. (ZeroTier Support and Release Signing Key) <contact@zerotier.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 74A5 E9C4 58E1 A431 F1DA  57A7 1657 1988 23E5 2A61

*** ZeroTier Service Quick Install for Unix-like Systems

*** Tested OSes / distributions:

***   MacOS (10.13+) (just installs ZeroTier One.pkg)
***   Debian Linux (7+)
***   RedHat/CentOS Linux (6+)
***   Fedora Linux (16+)
***   SuSE Linux (12+)
***   Mint Linux (18+)

*** Supported architectures vary by OS / distribution. We try to support
*** every system architecture supported by the target.

*** Please report problems to contact@zerotier.com and we will try to fix.

*** Detecting Linux Distribution

*** Found Debian, creating /etc/apt/sources.list.d/zerotier.list
Hit:1 http://download.zerotier.com/debian/bookworm bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Hit:5 https://deb.nodesource.com/node_20.x bookworm InRelease
Hit:6 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:7 https://repo.i2pd.xyz/debian bookworm InRelease
Hit:8 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gpg is already the newest version (2.2.40-1.1).
The following package was automatically installed and is no longer required:
  pi-language-support
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 16 not upgraded.

*** Installing zerotier-one package...
Hit:1 http://download.zerotier.com/debian/bookworm bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Hit:5 https://deb.nodesource.com/node_20.x bookworm InRelease
Hit:6 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:7 https://repo.i2pd.xyz/debian bookworm InRelease
Hit:8 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  pi-language-support
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  zerotier-one
0 upgraded, 1 newly installed, 0 to remove and 16 not upgraded.
Need to get 0 B/2,905 kB of archives.
After this operation, 9,956 kB of additional disk space will be used.
Selecting previously unselected package zerotier-one.
(Reading database ... 163212 files and directories currently installed.)
Preparing to unpack .../zerotier-one_1.12.2_arm64.deb ...
Unpacking zerotier-one (1.12.2) ...
Setting up zerotier-one (1.12.2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/zerotier-one.service → /lib/systemd/system/zerotier-one.service.
Processing triggers for man-db (2.11.2-2) ...

*** Enabling and starting ZeroTier service...
Synchronizing state of zerotier-one.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable zerotier-one

*** Waiting for identity generation...

*** Success! You are ZeroTier address [ *******].

# ZeroTier is now installed on your RaspiBlitz
# Joining zerotier network: *********
# OK - joined

rootzoll, 2024-03-19:
OK closing issue for now - please reopen/comment if error schows again.