raspiblitz/raspiblitz #4496: 1.11.0RC5 BTCpayserver won't start after updating (or uninstall/install)

kravens, 2024-03-23:
**Describe the bug**
Something is going wrong in the file permissions upon starting BTCpayserver, the installation goes fine with 0 issues.

**To Reproduce**
Install 1.11.0RC5 on Pi5
>sudo systemctl restart btcpayserver
See error

**Debug file**

```
Mar 23 17:19:13 raspberrypi systemd[1]: Started btcpayserver.service - BtcPayServer daemon.
Mar 23 17:19:14 raspberrypi dotnet[2355527]: info: Configuration:  Data Directory: /home/btcpay/.btcpayserver/Main
Mar 23 17:19:14 raspberrypi dotnet[2355527]: info: Configuration:  Configuration File: /home/btcpay/.btcpayserver/Main/settings.config
Mar 23 17:19:14 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Loading plugins from /home/btcpay/.btcpayserver/Plugins
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.Shopify - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.PointOfSale - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.PayButton - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.NFC - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.Crowdfund - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.Bitcoin - 1.12.5.0
Mar 23 17:19:15 raspberrypi dotnet[2355527]: info: Configuration:  Supported chains: BTC
Mar 23 17:19:16 raspberrypi dotnet[2355527]: info: Configuration:  BTC: Explorer url is http://127.0.0.1:24444/
Mar 23 17:19:16 raspberrypi dotnet[2355527]: info: Configuration:  BTC: Cookie file is /home/btcpay/.nbxplorer/Main/.cookie
Mar 23 17:19:16 raspberrypi dotnet[2355527]: info: Configuration:  Network: Mainnet
Mar 23 17:19:26 raspberrypi dotnet[2355527]: fail: BTCPayServer.Hosting.MigrationStartupTask: Error on the MigrationStartupTask
Mar 23 17:19:26 raspberrypi dotnet[2355527]: Npgsql.PostgresException (0x80004005): 42501: permission denied for schema public
Mar 23 17:19:26 raspberrypi dotnet[2355527]: POSITION: 14
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.Internal.NpgsqlConnector.ReadMessageLong(Boolean async, DataRowLoadingMode dataRowLoadingMode, Boolean readingNotifications, Boolean isReadingPrependedMessage)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16 token)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlDataReader.NextResult(Boolean async, Boolean isConsuming, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlDataReader.NextResult(Boolean async, Boolean isConsuming, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteReader(Boolean async, CommandBehavior behavior, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteReader(Boolean async, CommandBehavior behavior, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteNonQuery(Boolean async, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Migrations.Internal.Migrator.MigrateAsync(String targetMigration, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.EntityFrameworkCore.PostgreSQL.Migrations.Internal.NpgsqlMigrator.MigrateAsync(String targetMigration, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at BTCPayServer.Hosting.MigrationStartupTask.Migrate(CancellationToken cancellationToken) in /home/btcpay/btcpayserver/BTCPayServer/Hosting/MigrationStartupTask.cs:line 859
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at BTCPayServer.Hosting.MigrationStartupTask.ExecuteAsync(CancellationToken cancellationToken) in /home/btcpay/btcpayserver/BTCPayServer/Hosting/MigrationStartupTask.cs:line 85
Mar 23 17:19:26 raspberrypi dotnet[2355527]:   Exception data:
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Severity: ERROR
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     SqlState: 42501
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     MessageText: permission denied for schema public
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Position: 14
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     File: aclchk.c
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Line: 3650
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Routine: aclcheck_error
Mar 23 17:19:26 raspberrypi dotnet[2355527]: Unhandled exception. Npgsql.PostgresException (0x80004005): 42501: permission denied for schema public
Mar 23 17:19:26 raspberrypi dotnet[2355527]: POSITION: 14
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.Internal.NpgsqlConnector.ReadMessageLong(Boolean async, DataRowLoadingMode dataRowLoadingMode, Boolean readingNotifications, Boolean isReadingPrependedMessage)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16 token)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlDataReader.NextResult(Boolean async, Boolean isConsuming, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlDataReader.NextResult(Boolean async, Boolean isConsuming, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteReader(Boolean async, CommandBehavior behavior, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteReader(Boolean async, CommandBehavior behavior, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.NpgsqlCommand.ExecuteNonQuery(Boolean async, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteNonQueryAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Microsoft.EntityFrameworkCore.Migrations.Internal.Migrator.MigrateAsync(String targetMigration, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at Npgsql.EntityFrameworkCore.PostgreSQL.Migrations.Internal.NpgsqlMigrator.MigrateAsync(String targetMigration, CancellationToken cancellationToken)
Mar 23 17:19:26 raspberrypi dotnet[2355527]:    at BTCPayServer.Hosting.MigrationStartupTask.Migrate(CancellationToken cancellationToken) in /home/btcpay/btcpayserver/BTCPayServer/Hosting/MigrationStartupTask.cs:line 859
[... cut ...]
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Position: 14
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     File: aclchk.c
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Line: 3650
Mar 23 17:19:26 raspberrypi dotnet[2355527]:     Routine: aclcheck_error
Mar 23 17:19:26 raspberrypi systemd[1]: btcpayserver.service: Main process exited, code=exited, status=134/n/a
Mar 23 17:19:26 raspberrypi systemd[1]: btcpayserver.service: Failed with result 'exit-code'.
Mar 23 17:19:26 raspberrypi systemd[1]: btcpayserver.service: Consumed 10.086s CPU time.
```

rootzoll, 2024-03-24:
Basic Install looks fine in v1.11.0rc5 - BTCPayServer starts and I can crfeate Login & Store. 

@kravens can you describe your setup/history of your BTCPayServer so that we can try to reproduce?

kravens, 2024-03-24:
I've had it working at the start just after the install, and then updated to the 1.11 repo to fix some of the other services.
Afterwards I'm seeing the nginx gateway error, turned out BTCpayserver is no longer starting anymore :(
I already had a store on the data SSD from previous install, that it tried to recover.

openoms, 2024-03-25:
@kravens did you reflash with the v011.0rc5 image pr just updated the scripts?

kravens, 2024-03-25:
> @kravens did you reflash with the v011.0rc5 image pr just updated the scripts?

I did not reflash from RC4 I believe, I updated from a flashed RC4 to RC5 via the Patch menu.
Probably a good idea to reflash RC5 once too if that could be the issue.

> 
> Try running:
> 
> ```
> sudo -u postgres psql -d btcpaymainnet -c "GRANT ALL PRIVILEGES ON SCHEMA public TO btcpay;"
> ```
> 
> we could switch to running this always for old postgres btcpay installs.

This results in the following:"
ERROR:  could not read block 1 in file "global/1214": read only 0 of 8192 bytes

> 
> Try this snippet which could be part of the PR:
> 
> ```
>   # Check if the btcpay user has ALL privileges on the public schema. If the permissions are not set to true ('t'), grant them
>   if [ "$(sudo -u postgres psql -d btcpaymainnet -t -c "SELECT has_schema_privilege('btcpay', 'public', 'ALL');" | xargs)" = "t" ]; then
>     echo "# Permissions are correctly set. No action needed."
>   else
>     echo "# Correct permissions not found. Setting them now."
>     sudo -u postgres psql -d btcpaymainnet -c "GRANT ALL PRIVILEGES ON SCHEMA public TO btcpay;"
>   fi
> ```
Running this snippet also gives an error:
ERROR:  unrecognized privilege type: "ALL"
# Correct permissions not found. Setting them now.
ERROR:  could not read block 1 in file "global/1214": read only 0 of 8192 bytes

openoms, 2024-03-25:
RC4 had the fix also, so that shouldn't matter. On Raspiblitz version was the BTCPay database created?

Can you try uninstalling BTCPay including deleting it's data? I suspect there is something corrupt in postgres and it should be ok on a new install.
I did install BTCPay a few times recently and had nothing like this.

kravens, 2024-03-25:
The database must be from the 1.10 release that I migrated from recently.

kravens, 2024-03-25:
During my fresh install of the fatpack RC5 image I also noticed an error for Postgres that said: Quitfinished or something like that on the LCD screen.
Same issue persists after the fresh install, so I think it must be related to the BTCPay data on my SSD.
I'll try to delete everything and install BTCPay from scratch.

kravens, 2024-03-25:
> admin@192.168.1.108:/mnt/hdd/app-data/postgresql ₿ ls -lh
> total 8.0K
> drwxr-xr-x 3 postgres postgres 4.0K May  5  2023 13
> drwxr-xr-x 3 postgres postgres 4.0K Mar 20 14:02 15

These folders are in my postgres app-data, maybe one of them is corrupted?

kravens, 2024-03-25:
Looks like my issue is indeed only the Postgresql install that is failing on my fresh RC5 install now:
```
admin@192.168.1.108:~/config.scripts ₿ ./bonus.postgresql.sh on
# Install PostgreSQL
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  postgresql-doc
The following NEW packages will be installed:
  postgresql
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 0 B/10.1 kB of archives.
After this operation, 15.4 kB of additional disk space will be used.
Selecting previously unselected package postgresql.
(Reading database ... 145503 files and directories currently installed.)
Preparing to unpack .../postgresql_15+248_all.deb ...
Unpacking postgresql (15+248) ...
Setting up postgresql (15+248) ...
# Good link in /var/lib/postgresql
Synchronizing state of postgresql.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable postgresql
Created symlink /etc/systemd/system/multi-user.target.wants/postgresql.service → /lib/systemd/system/postgresql.service.
psql (PostgreSQL) 15.6 (Debian 15.6-0+deb12u1)
sleep 1/30
sleep 2/30
sleep 3/30
sleep 4/30
sleep 5/30
sleep 6/30
sleep 7/30
sleep 8/30
sleep 9/30
sleep 10/30
sleep 11/30
sleep 12/30
sleep 13/30
sleep 14/30
sleep 15/30
sleep 16/30
sleep 17/30
sleep 18/30
sleep 19/30
sleep 20/30
sleep 21/30
sleep 22/30
sleep 23/30
sleep 24/30
sleep 25/30
sleep 26/30
sleep 27/30
sleep 28/30
sleep 29/30
sleep 30/30
● postgresql.service - PostgreSQL RDBMS
     Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2024-03-25 22:04:52 GMT; 30s ago
    Process: 305069 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
   Main PID: 305069 (code=exited, status=0/SUCCESS)
        CPU: 1ms

Mar 25 22:04:52 raspberrypi systemd[1]: Starting postgresql.service - PostgreSQL RDBMS...
Mar 25 22:04:52 raspberrypi systemd[1]: Finished postgresql.service - PostgreSQL RDBMS.
FAIL - Was not able to start PostgreSQL service
```

kravens, 2024-03-25:
Extra debug logs here: https://termbin.com/u04u

rootzoll, 2024-03-26:
@openoms is this something to wait for solving for v1.11.0 or something individual that could be pushed to v1.11.1 if needed?

openoms, 2024-03-26:
@rootzoll couldn't reproduce this yet, so  I suspect it is an individual issue

@kravens can you try to reinstall postgres with:
```
config.scripts/bonus.postgresql.sh off
config.scripts/bonus.postgresql.sh on
```

do you get any errors?

kravens, 2024-03-26:
I had already done the reflashing of RC5 on the microSD card and this issue persisted, the logs above are the result of the bonus.postgresql.sh off/on action too.

openoms, 2024-03-26:
@kravens will need to check the error logs of postgres to find out what is wring with it.

```
sudo journalctl -fu postgresql -n100
```
+ whichever other service names it has

kravens, 2024-03-26:
This is the journalctl log that I get:

```
Mar 25 22:34:08 raspberrypi systemd[1]: Finished postgresql.service - PostgreSQL RDBMS.
Mar 26 20:29:12 raspberrypi systemd[1]: postgresql.service: Deactivated successfully.
Mar 26 20:29:12 raspberrypi systemd[1]: Stopped postgresql.service - PostgreSQL RDBMS.
Mar 26 20:29:51 raspberrypi systemd[1]: Starting postgresql.service - PostgreSQL RDBMS...
Mar 26 20:29:51 raspberrypi systemd[1]: Finished postgresql.service - PostgreSQL RDBMS.
Mar 26 21:52:59 raspberrypi systemd[1]: postgresql.service: Deactivated successfully.
Mar 26 21:52:59 raspberrypi systemd[1]: Stopped postgresql.service - PostgreSQL RDBMS.
Mar 26 21:52:59 raspberrypi systemd[1]: Stopping postgresql.service - PostgreSQL RDBMS...
Mar 26 21:52:59 raspberrypi systemd[1]: Starting postgresql.service - PostgreSQL RDBMS...
Mar 26 21:52:59 raspberrypi systemd[1]: Finished postgresql.service - PostgreSQL RDBMS.
```

openoms, 2024-03-27:
@kravens got this reproduced.

Comparing the databases between my rc4 update to RC5:
```
sudo -u postgres psql -d btcpaymainnet -l
                                                     List of databases
       Name       |  Owner   | Encoding |   Collate   |    Ctype    | ICU Locale | Locale Provider |   Access privileges   
------------------+----------+----------+-------------+-------------+------------+-----------------+-----------------------
 btcpaymainnet    | postgres | UTF8     | C           | C           |            | libc            | 
 lnbits_db        | postgres | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |            | libc            | 
 nbxplorermainnet | postgres | UTF8     | C           | C           |            | libc            | 
 postgres         | postgres | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |            | libc            | 
 template0        | postgres | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |            | libc            | =c/postgres          +
                  |          |          |             |             |            |                 | postgres=CTc/postgres
 template1        | postgres | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 |            | libc            | =c/postgres          +
                  |          |          |             |             |            |                 | postgres=CTc/postgres
(6 rows)
```
vs a working instance:
```
       Name       |  Owner   | Encoding |   Collate   |    Ctype    | ICU Locale | Locale Provider |    Access privileges     
------------------+----------+----------+-------------+-------------+------------+-----------------+--------------------------
 btcpaymainnet    | postgres | UTF8     | en_US.UTF-8 | en_US.UTF-8 |            | libc            | =Tc/postgres            +
                  |          |          |             |             |            |                 | postgres=CTc/postgres   +
                  |          |          |             |             |            |                 | btcpay=CTc/postgres
 lnbits_db        | postgres | UTF8     | en_US.UTF-8 | en_US.UTF-8 |            | libc            | =Tc/postgres            +
                  |          |          |             |             |            |                 | postgres=CTc/postgres   +
                  |          |          |             |             |            |                 | lnbits_user=CTc/postgres
 nbxplorermainnet | postgres | UTF8     | C           | C           |            | libc            | =Tc/postgres            +
                  |          |          |             |             |            |                 | postgres=CTc/postgres   +
                  |          |          |             |             |            |                 | nbxplorer=CTc/postgres
 postgres         | postgres | UTF8     | en_US.UTF-8 | en_US.UTF-8 |            | libc            | 
 template0        | postgres | UTF8     | en_US.UTF-8 | en_US.UTF-8 |            | libc            | =c/postgres             +
                  |          |          |             |             |            |                 | postgres=CTc/postgres
 template1        | postgres | UTF8     | en_US.UTF-8 | en_US.UTF-8 |            | libc            | =c/postgres             +
                  |          |          |             |             |            |                 | postgres=CTc/postgres
(6 rows)
```

kravens, 2024-03-27:
Nice work! How do we fix the access privilege mismatch in the updated RC4?

openoms, 2024-03-27:
it looks like even the new install has trouble writing the databases
```
# Generate the database for nbxplorer                                                                                                       
CREATE DATABASE                                                                                                                             
CREATE ROLE                                                                                                                                 
ERROR:  could not read block 1 in file "global/1214": read only 0 of 8192 bytes                                                             
ERROR:  could not read block 0 in file "global/1214": read only 0 of 8192 bytes    
```

```
# Generate the database for btcpay                                                                                                          
CREATE DATABASE                                                                                                                             
CREATE ROLE                                                                                                                                 
ERROR:  could not read block 1 in file "global/1214": read only 0 of 8192 bytes                                                             
ERROR:  could not read block 0 in file "global/1214": read only 0 of 8192 bytes   
```
The files are symlinked from the disk, but it was working before, not sure why it is created as corrupt somehow.

I see this on my RPi5 + NVMe, is that your setup also @kravens?

Can't run this run right now on an RPi4, but would be good to know if is somehow related to the NVMe ?

kravens, 2024-03-27:
@openoms I also run a Pi5 with 2TB NVMe on a bottom hat.

openoms, 2024-03-27:
> I also run a Pi5 with 2TB NVMe on a bottom hat.

something is fishy on my RPi5, will test on an RPi4 tomorrow. 
Now getting:
```
sudo smartctl -a /dev/nvme0n1p1
smartctl 7.3 2022-02-28 r5338 [aarch64-linux-6.6.20+rpt-rpi-v8] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

Read NVMe Identify Controller failed: NVME_IOCTL_ADMIN_CMD: Input/output error
```

kravens, 2024-03-27:
@openoms I'm using only ethernet cable, I thought that I had wifi and bt disabled in the /boot/config.txt... but now I'm checking it turns out that file is not used and in /boot/firmware/config.txt I'm not explicitly disabling wifi or bluetooth!
I'll disable them and do a reboot to check what is going on then.

kravens, 2024-03-27:
`sudo smartctl -a /dev/nvme0n1p1
smartctl 7.3 2022-02-28 r5338 [aarch64-linux-6.6.20+rpt-rpi-v8] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       Lexar SSD NM620 2TB
Serial Number:                      NHN986R008058P1110
Firmware Version:                   9846
PCI Vendor/Subsystem ID:            0x1e4b
IEEE OUI Identifier:                0xcaf25b
Total NVM Capacity:                 2,048,408,248,320 [2.04 TB]
Unallocated NVM Capacity:           0
Controller ID:                      0
NVMe Version:                       1.4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          2,048,408,248,320 [2.04 TB]
Namespace 1 Utilization:            233,155,514,880 [233 GB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            caf25b 038000035e
Local Time is:                      Wed Mar 27 22:33:49 2024 GMT
Firmware Updates (0x14):            2 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x001f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat
Log Page Attributes (0x02):         Cmd_Eff_Lg
Maximum Data Transfer Size:         128 Pages
Warning  Comp. Temp. Threshold:     90 Celsius
Critical Comp. Temp. Threshold:     95 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     6.50W       -        -    0  0  0  0        0       0
 1 +     5.80W       -        -    1  1  1  1        0       0
 2 +     3.60W       -        -    2  2  2  2        0       0
 3 -   0.7460W       -        -    3  3  3  3     5000   10000
 4 -   0.7260W       -        -    4  4  4  4     8000   45000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        35 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    28,279,586 [14.4 TB]
Data Units Written:                 10,788,354 [5.52 TB]
Host Read Commands:                 159,726,183
Host Write Commands:                91,054,411
Controller Busy Time:               1,073
Power Cycles:                       8
Power On Hours:                     1,509
Unsafe Shutdowns:                   5
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               35 Celsius
Temperature Sensor 2:               34 Celsius

Error Information (NVMe Log 0x01, 16 of 64 entries)
No Errors Logged
`

openoms, 2024-03-28:
testing now after applied rfkill:
```
sudo rfkill block wifi
sudo rfkill block bluetooth
```
```
 ₿ sudo rfkill list
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: yes
        Hard blocked: no
```

Will need to check if wifi is soft blocked by default. @kravens did you ever have the wifi activated or just used cable only on this instance (including if it was migrated from other hardware) ?

kravens, 2024-03-28:
I believe this node only used cable in the last 2 years. In very early configuration 2 hardware migrations before I might have used wifi with it that was later disabled in config file.

openoms, 2024-03-28:
@kravens not sure if wifi plays a role, might have a role in the corruption.

Unfortunately only nuking the whole postgres data have fixed the problem for me. Hope @kravens you did not have anything irreplaceable in btcpay as that database looks corrupted and probably needs a lot of effort to recover. 

If you have LNbits or anything else using postgres make MIGRATION file backup and try the individual LNbits postgres backup from the menu (taht might not work as postgres is not running).

If you ok with starting a  new one could test the fixed script from the PR:
```
# download
wget -O ~/config.scripts/bonus.postgresql.sh  ~/config.scripts/bonuhttps://raw.githubusercontent.com/raspiblitz/raspiblitz/b1c2e95ff54c094b2faaad2da8c8cdf1c29537fc/home.admin/config.scripts/bonus.postgresql.sh 

~/config.scripts/bonus.postgresql.sh off
```

**take extra care and only use this on a test machine with no valuable data in postgres**
!!!This deletes all postgres databases, not just btcpayserver (remove the # from front)
```
# sudo pg_dropcluster 15 main
```
```
# install postgres again
~/config.scripts/bonus.postgresql.sh on
```
Now can reinstall btcpayserver to start clean.

openoms, 2024-03-28:
The fix for recovery was #4506

kravens, 2024-03-28:
`Warning: corrupted cluster: data directory does not exist`

Looks like my own attempt at removing the postgres database made things worse xD.

kravens, 2024-03-28:
[image]
After dropping the postgres database and installing postgresql again this is the next error I run into:
```
Mar 28 19:01:18 raspberrypi systemd[1]: Started btcpayserver.service - BtcPayServer daemon.
Mar 28 19:01:19 raspberrypi dotnet[4086013]: Unhandled exception: System.ComponentModel.Win32Exception (2): An error occurred trying to start process '/home/btcpay/btcpayserver/BTCPayServer/bin/Release/net8.0/BTCPayServer' with working directory '/home/btcpay/btcpayserver/BTCPayServer'. No such file or directory
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at System.Diagnostics.Process.ForkAndExecProcess(ProcessStartInfo startInfo, String resolvedFilename, String[] argv, String[] envp, String cwd, Boolean setCredentials, UInt32 userId, UInt32 groupId, UInt32[] groups, Int32& stdinFd, Int32& stdoutFd, Int32& stderrFd, Boolean usesTerminal, Boolean throwOnNoExec)
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at System.Diagnostics.Process.StartCore(ProcessStartInfo startInfo)
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at Microsoft.DotNet.Cli.Utils.Command.Execute(Action`1 processStarted)
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at Microsoft.DotNet.Tools.Run.RunCommand.Execute()
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at System.CommandLine.Invocation.InvocationPipeline.Invoke(ParseResult parseResult)
Mar 28 19:01:19 raspberrypi dotnet[4086013]:    at Microsoft.DotNet.Cli.Program.ProcessArgs(String[] args, TimeSpan startupTime, ITelemetry telemetryClient)
Mar 28 19:01:19 raspberrypi systemd[1]: btcpayserver.service: Main process exited, code=exited, status=1/FAILURE
Mar 28 19:01:19 raspberrypi systemd[1]: btcpayserver.service: Failed with result 'exit-code'.
```

openoms, 2024-03-29:
The underlying problem was that postgres was installed and ran as an optional CLN dependency which created a new postgres cluster.

Reinstalling BTCPay ran the `bonus.postgresql.sh on` which restored the old cluster from the disk which now appeared corrupt.

Removing the corrupt cluster and reinstalling postgres with only the script solved the issue.

If you still have a copy of the 
`/mnt/hdd/app-data/postgres` dir (in a MIGRATION file for example) could copy it in in place on the disk and should be able to recover from that using the final v0.11.0 release image (expected tomorrow).

If have no old copy of the data might need to move the corrupt postgres directory away anyway to be able to start clean with the next refresh.

To move the old dir just run:
```
sudo systemctl stop postgresql
sudo mv /mnt/hdd/app-data/postgres `/mnt/hdd/app-data/postgres.backup
```
and refresh with the final image.

kravens, 2024-03-29:
I'll probably do a reflash with release 1.11 after removing the postgres database, I've been messing up scripts a lot to try and get BTCPay to run (which didn't work).

openoms, 2024-03-30:
Ok, at last this is looking good, but needed to get out of the mess created as described above. 
@kravens you will also need to delete the postgres directory from the disk if it was corrupted - I had the issue that I couldn;t restore it because was attempting to restore the corrupted db.
The best would be to make it work on your current setup so it should be restored ok with the fixed script (which does not install postgres for CLN - so there is no cluster created separately.)

Here are the steps to clean properly and make it work:

**WARNING THESE ARE TO CLEAN AN EXISTING CORRUPTED POSTGRES DB COMPLETELY**
```
# delete pg symlink
sudo rm -f /var/lib/postgresql

# delete old database on disk
sudo rm /mnt/hdd/app-data/postgresql

openoms, 2024-03-30:
One additional issue can be that first we had the postgres data directory in `/mnt/hdd/app-data/postgresql/13/main` and now as there is a higher version used it is in `/mnt/hdd/app-data/postgresql/15/main`.
Need to find something out to ensure we can recover the old data even between versions.

kravens, 2024-03-30:
I indeed had both folders, 13 and 15 on my data drive.
I've successfully installed BTCpayserver 1.12.5 but it still crashes when launching somehow... so likely a fresh install of the sd-card is needed.

openoms, 2024-03-30:
There are some extra steps here we did not consider so far: https://chat.openai.com/share/1fff0f33-ab6f-44fc-ba78-1c505a9ea743

looking into `pg_upgrade` or if we should keep using postgres 13

openoms, 2024-04-04:
postgres was recovered and running from prev pg 15 data, btcpay is working ok

kravens, 2024-04-10:
[image]
[image]
Installed RC7 and my new BTCPayserver (v1.12.5) is running smoothly again! Only encountered one error with postgresql@13-main.service during the install, likely not critical.

openoms, 2024-04-10:
yes, correct postgres 13 doesn't need to work at that point so it is not a problem. 

Thank you very much for reporting this issue and testing here @kravens !

PR #4504 (2024-03-28): use the installed postgres version to create the cluster
attempt to fix #4496

needs testing with recovery and new install

building a lean image from the branch in https://github.com/raspiblitz/raspiblitz/actions/runs/8458921265