raspiblitz/raspiblitz #4497: BUILD FAILED --> PGP Verify not OK

M2402, 2024-03-24:
**Describe the bug**
Installing on a Intel NUC in a Proxmox environment, Debian 11.7.
Following https://cercatrova.blog/raspiblitz-auf-proxmox-installieren/

End of output... (see complete output in attached file)
-----------------------------------------------------
goodSignature(0)
correctKey(0)
# BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)
-----------------------------------------------------

**To Reproduce**
tried multiple VMs, same result

**Debug file**
[build card output.txt](https://github.com/raspiblitz/raspiblitz/files/14736510/build.card.output.txt)

rootzoll, 2024-03-25:
Marked for inspection on next Proxmox testing round

openoms, 2024-03-25:
Switch to a US English environment to solve this.

rootzoll, 2024-09-12:
OK tested again with the lastest build script in `dev` branch and it works in my Proxmox environment. Closing issue.