raspiblitz/raspiblitz #4517: v1.11.0rc6 Hangs on Shutdown

rootzoll, 2024-04-03:
Happens on RP4 with min & fat image of v1.11.rc6

[image]

rootzoll, 2024-04-03:
there is also a reboot happening even before the boostrap script starts (because stop flag gets ignored). Very tricky on a min-image where you dont see anything on the LCD and the it hangs on reboot and not responding to SSH.

rootzoll, 2024-04-03:
It looks like it does not hang when the sd card gets build on a real RaspberryPi and not in packer.

openoms, 2024-04-04:
The latest image had no problems on restart during recovery on RPi4 and RPi5

rootzoll, 2024-04-05:
The latest image rc6 contain a fix that needs to be tested more.

rootzoll, 2024-04-16:
Looks good now - closing for v1.11.0 release

closed 2024-04-16 by rootzoll