raspiblitz/raspiblitz #4534: Hangs on shutdown/restart

rootzoll, 2024-04-07:
This is a sometimes happening error that the RaspiBlitz hangs in shutdown with the message that watchdog cannot stop.

This bug is under investigation and we need your help how to reproduce this problem to fix it. 

So if you experience it, please report:
- hat sd card image did you use (version, release candidate, min or fatpack)
- in what state and on which action dis the reboot/shutdown happen (during setup, after setup, etc)
- and what bonus apps do you have installed.

rootzoll, 2024-04-07:
Just for deeper research there are the running services on min & fatpack before setup to compare:

v1.11.0rc6-min:`systemctl list-units --type=service --state=running`
```
  UNIT                      LOAD   ACTIVE SUB     DESCRIPTION
  avahi-daemon.service      loaded active running Avahi mDNS/DNS-SD Stack
  cron.service              loaded active running Regular background program processing daemon
  dbus.service              loaded active running D-Bus System Message Bus
  fail2ban.service          loaded active running Fail2Ban Service
  getty@tty1.service        loaded active running Getty on tty1
  i2pd.service              loaded active running I2P Router written in C++
  ModemManager.service      loaded active running Modem Manager
  NetworkManager.service    loaded active running Network Manager
  nginx.service             loaded active running A high performance web server and a reverse proxy server
  polkit.service            loaded active running Authorization Manager
  redis-server.service      loaded active running Advanced key-value store
  rsyslog.service           loaded active running System Logging Service
  rtkit-daemon.service      loaded active running RealtimeKit Scheduling Policy Service
  smartmontools.service     loaded active running Self Monitoring and Reporting Technology (SMART) Daemon
  ssh.service               loaded active running OpenBSD Secure Shell server
  systemd-journald.service  loaded active running Journal Service
  systemd-logind.service    loaded active running User Login Management
  systemd-timesyncd.service loaded active running Network Time Synchronization
  systemd-udevd.service     loaded active running Rule-based Manager for Device Events and Files
  tor@default.service       loaded active running Anonymizing overlay network for TCP
  triggerhappy.service      loaded active running triggerhappy global hotkey daemon
  user@1000.service         loaded active running User Manager for UID 1000
  user@1001.service         loaded active running User Manager for UID 1001
  vnstat.service            loaded active running vnStat network traffic monitor
  wpa_supplicant.service    loaded active running WPA supplicant
```

v1.11.0rc6-fat:`systemctl list-units --type=service --state=running`
```
  UNIT                          LOAD   ACTIVE SUB     DESCRIPTION
  avahi-daemon.service          loaded active running Avahi mDNS/DNS-SD Stack
  blitzapi.service              loaded active running BlitzBackendAPI
  cron.service                  loaded active running Regular background program processing daemon
  dbus.service                  loaded active running D-Bus System Message Bus
  fail2ban.service              loaded active running Fail2Ban Service
  getty@tty1.service            loaded active running Getty on tty1
  i2pd.service                  loaded active running I2P Router written in C++
  ModemManager.service          loaded active running Modem Manager
  NetworkManager.service        loaded active running Network Manager
  nginx.service                 loaded active running A high performance web server and a reverse proxy server
  polkit.service                loaded active running Authorization Manager
  redis-server.service          loaded active running Advanced key-value store
  rsyslog.service               loaded active running System Logging Service
  rtkit-daemon.service          loaded active running RealtimeKit Scheduling Policy Service
  serial-getty@ttyAMA10.service loaded active running Serial Getty on ttyAMA10
  smartmontools.service         loaded active running Self Monitoring and Reporting Technology (SMART) Daemon
  ssh.service                   loaded active running OpenBSD Secure Shell server
  systemd-journald.service      loaded active running Journal Service
  systemd-logind.service        loaded active running User Login Management
  systemd-timesyncd.service     loaded active running Network Time Synchronization
  systemd-udevd.service         loaded active running Rule-based Manager for Device Events and Files
  tor@default.service           loaded active running Anonymizing overlay network for TCP
  triggerhappy.service          loaded active running triggerhappy global hotkey daemon
  user@1000.service             loaded active running User Manager for UID 1000
  user@1001.service             loaded active running User Manager for UID 1001
  vnstat.service                loaded active running vnStat network traffic monitor
```

rootzoll, 2024-04-07:
Something to try out --> `sudo nano /etc/systemd/system.conf` to activate the option `RuntimeWatchdogSec=2min`

here are some details on this option:
```
Description: This setting specifies the timeout for the reboot watchdog. If a reboot takes longer than the specified time, the system will be hard-rebooted. This is useful for ensuring that the system recovers from a state where it has begun the reboot process but gets stuck before completion.

Usage: Set to a time value, such as 10min. If a reboot process exceeds this duration, the watchdog triggers a system reboot to recover from potential hang-ups during shutdown or reboot sequences.
```

rootzoll, 2024-04-16:
so far rc7 stable ... closing for final release

openoms, 2024-04-24:
Reopening as this is still happening occasionally eg https://t.me/raspiblitz/142982 + reported by @fusion44 
Some ideas:

for the watchdog problem can try reducing the TimeoutStopSec in:
```
# An extended timeout period is needed to allow for database compaction
# and other time intensive operations during startup. We also extend the
# stop timeout to ensure graceful shutdowns of lnd.
TimeoutStartSec=1200
TimeoutStopSec=3600
```
with the command:
```
sudo systemctl edit --full lnd
```

the watchdog service is set to:
RuntimeWatchdogSec=600s in  /etc/systemd/system.conf

I feel that should be closer to 3600 if we wan to keep patient with LND

patricklocher, 2024-05-03:
Raspberry Pi 5 Model B Rev 1.0 is rebooting as expected on 1.11.0
Raspberry Pi 4 Model B Rev 1.5 got this issue when rebooting till update to 1.11.0

bitsam, 2024-08-26:
The watchdog hung error just happened to me on the first boot of upgrading from v1.11.0 to v1.11.2 with the fatpack image. I have a rpi4.

rootzoll, 2024-08-30:
@bitsam with RaspiBlitz v1.12.0 we will change to a new RaspiOS base image - lets keep fingers crossed that this issues goes away with that update for the rpi4s

rootzoll, 2024-11-12:
TODO: Because we cannot garantuee that a future RaspiOS update will fix this - for v1.12.0 we should at least change the instruction text given on a reboot (when its a RP4) to someting like "if you see Watchdog failes on LCD or you waited at least 10min, the disconnect & reconnect power". Would also need this for WebUI is possible.

woiwoiwoiwoi, 2024-12-13:
After today's `sudo apt update && upgrade` my RB rebooted correctly for the first time in a long while. I'm still on v1.11.2.

rootzoll, 2024-12-13:
> After today's sudo apt update && upgrade my RB rebooted correctly for the first time in a long while. I'm still on v1.11.2.

So lets test for v1.12.0 if we get rid of this problem, where we run on a new base image and run a full `apt upgrade`:
https://github.com/rootzoll/raspiblitz/blob/dev/build_sdcard.sh#L354 - that would be great.

woiwoiwoiwoi, 2025-01-15:
I upgraded to 1.11.4 and reboot does not work anymore, even after `sudo apt update && upgrade`.

rootzoll, 2025-02-03:
> I upgraded to 1.11.4 and reboot does not work anymore, even after sudo apt update && upgrade.

And you end up with the "watchdog" error?

woiwoiwoiwoi, 2025-02-03:
> And you end up with the "watchdog" error?

Yes.

I've stumbled into two other problems after the update though: [high CPU load due to mempool.space](https://github.com/raspiblitz/raspiblitz/issues/4936) (maybe you have some input here?) and Specter Desktop not working (haven't created an issue). So I'm considering setting everything up from scratch again. I would prefer not to have to do that - but if I do, then I would check for this restarting issue again.

rootzoll, 2025-02-03:
> Specter Desktop not working (haven't created an issue)

If it looks like a general problem, please open an issue on that too.

WatchDog TODO: Test if with the latest RaspiOS the watchdog still happens.

rootzoll, 2025-05-17:
Updated to latest RaspiOS for v1.12.0 .. looks good. Pushing to v1.12.1 to recheck if it was stil happening.