raspiblitz/raspiblitz #4540: Sign from a new Adress (Pocket Bitcoin Buy Helper)

bvolution, 2024-04-16:
Hi, 

## Purpose
from time to time I need to sign a message from a specific adress.
E.g. for the case when one wants to buy bitcoin via PocketBitcoin App and needs
to verify ownership via signing a message from a specific address.

I developed a small handy script that helps with this. 

## Overview / Summary

- Get address (either input manually and verify or generate a new one
- Enter message
- Returns the signature

## Suggestion

This might be useful for other people, it might could be added with an alias such as `sign` or `pocket`

```sh
#!/bin/bash

# 1. Ask if a new address should be generated or an existing one should be entered
read -p "Do you want to generate a new address? (y/n) " generate_new

# 1.a. If generating a new address, ask for the address type
if [ "$generate_new" == "y" ]; then
    echo "Generate a wallet new address. Address-types has to be one of:"
    echo "1. p2wkh:  Pay to witness key hash"
    echo "2. np2wkh: Pay to nested witness key hash"
    echo "3. p2tr:   Pay to taproot pubkey"
    read -p "Enter the address type (1-3 or string): " address_type
    case "$address_type" in
        1|"p2wkh")
            address_type="p2wkh"
            ;;
        2|"np2wkh")
            address_type="np2wkh"
            ;;
        3|"p2tr")
            address_type="p2tr"
            ;;
        *)
            echo "Error: Invalid address type."
            exit 1
            ;;
    esac
    address=$(lncli newaddress $address_type)
    address_variable=$(echo $address | jq -r '.address')
else
    # 1.b. Check if the manually entered address is valid
    read -p "Enter the existing address: " address
    if ! lncli validateaddress "$address" | grep -q "isvalid: true"; then
        echo "Error: The entered address is not valid."
        exit 1
    fi
    address_variable=$address
fi

# 2. Ask for the message to sign and save it to a variable
read -p "Enter the message to sign: " message_to_sign

# 3. Execute the lncli wallet addresses signmessage command
signature_js=$(lncli wallet addresses signmessage --address $address_variable --msg "$message_to_sign")
signature=$(echo $signature_js | jq -r '.signature')
echo "The address is: $address_variable"
echo "The message to sign is: $message_to_sign"
echo "The signature is: $signature"
```

rootzoll, 2024-05-24:
OK I added a PR with this script. see #4582

rootzoll, 2024-05-28:
PR merged ... will be part of RaspiBlitz v1.11.1rc1

closed 2024-07-08 by rootzoll