raspiblitz/raspiblitz #4543: 1.11.0 Lean Image did not start setup process

neversleep404, 2024-04-18:
HW:
Geekworm X1001 PCIe to M.2 NVMe
WD_BLACK SN770 NVMe SSD 2 TB
ELEGOO Display 3.5" Zoll TFT LCD

What happened: Insert the fresh burned SD-Card. Power on. Fan is loud. Screen stays white. SSH did also not work. SD-Card light flashes. Nothing happens.

Manual Power off. Manual Power on. Fan is loud. system boots. Screen stays white. 3 or 4 automatic restarts. Screen is on. Setup Process Starts.

rootzoll, 2024-05-24:
Did you had any luck getting it running.

I tested the Geekworm X1001 once and it worked. Maybe its something with the NVMe?
Did the NVMe worked with other adapters?

neversleep404, 2024-05-25:
Welches Build soll ich testen?

neversleep404, 2024-05-25:
Which build should I test?

rootzoll, 2024-05-25:
Best guess its that its something with the NVMe - do you have another NVMe to test with .. even its its a much smaller one.

neversleep404, 2024-05-25:
HW:
**Geekworm X1002** PCIe to M.2 NVMe
WD_BLACK SN770 NVMe SSD 2 TB
ELEGOO Display 3.5" Zoll TFT LCD

Insert the fresh burned SD-Card. Power on. 2 or 3 automatic restarts and it works. 

My X1001 is running fine and productive with the FatPack. --> don't change a running system

closed 2024-06-09 by rootzoll