raspiblitz/raspiblitz #4553: [RP5] Update EEPROM latest Firmware

rootzoll, 2024-04-23:
ATM v1.11.0 updates EEPROM in bootstrap to  build 1701887365  - see
https://github.com/raspiblitz/raspiblitz/blob/58d568685ecd31e7ccbcebf6d4eb8cbcebcbce23/home.admin/_bootstrap.sh#L512

Should force an update the the latest firmware version with v1.11.1 - testers confirm that it improves NVMe stability. Could make sense in preperation towards v1.12.0

openoms, 2024-04-24:
To update manually for testing in the meantime:
```
sudo apt-get update
sudo apt-get upgrade -y
sudo apt-get install -y rpi-eeprom
sudo rpi-eeprom-update -a
```

rootzoll, 2024-05-24:
The PR will force a firmware update to 1708097321 from Feb24 - even while there is already a firmware 1713610410 from Apr24 on my other machine that lastest one will not be shown as "LATEST" and so to prevent forcing the Pis into a Updateloop I think its OK to just force update if its older then Feb24 firmware.

rootzoll, 2024-05-24:
I opened up an issue #4584 for further update towards RaspiBlitz v1.12.0

openoms, 2024-05-25:
Already running my RPi5 on 1713610410 without issues.

PR #4583 (2024-05-25): #4553 Update EEPROM firmware 1708097321
fixes #4553 - tested on my machine
closed 2024-05-25 by rootzoll