raspiblitz/raspiblitz #4554: Raspiblitz-Screen black

Sigevi, 2024-04-24:
The Rapsiblitz LCD stays black after final reboot. With ssh it's possible to check out the normal view and everything else. 
I tried to update it again but the same Result came out. 

It worked well with the old version. 

Thanks!

milezzz, 2024-04-27:
Same issue for me after upgrading to v1.11 (fat). For me it's a black screen with blinking cursor in top corner. I know the screen is working because if I reflash the SDcard I can see proper output to the screen until the final reboot after its setup process.

I tried reloading the driver with ```LCD-show```, also tried from cli: ```lcd```

This is a RaspiBlitz running since 2019 without any issues.

milezzz, 2024-04-27:
Please disregard my post above, I had the screen settings incorrect. 
[image]

openoms, 2024-05-31:
Correct, activating the Touchscreen does result in a black screen currently. Should be deactivated until fixed.

PR #4609 (2024-06-09): #4554 Deactivate Touchscreen Option in SSH Menu
fixes: #4554
closed 2024-06-09 by rootzoll