raspiblitz/raspiblitz #4555: Stuck at "Starting RaspiBlitz. Please wait ..." with amd64-lean-legacyboot-image-build 1.11

trumfnator, 2024-04-25:
I was successfully using a 1.11rc0 amd64-lean-legacyboot build before.

I flashed the image like described in https://github.com/raspiblitz/raspiblitz/tree/v1.11/ci

[image]

Stuck at this screen after flashing amd64-lean-legacy-image-build 1.11. on existing 1.10 build (also amd64 lean legacy). Before this screen, there was „background.scan not ready yet ... (please wait)“ repeatably in the terminal.

lsblk shows:
[image]

df shows:
[image]

Successfully added hdd to mnt/hdd/bitcoin manually and also updated fstab.

After reboot, I'm still stuck with "background.scan not ready yet ...(please wait)" and the "Starting Raspiblitz Please Wait ..." Screen.

openoms, 2024-04-28:
@trumfnator can you please check the output of:
```
sudo /home/admin/config.scripts/blitz.datadrive.sh status
```
It should have:
```
hddRaspiData=1
```
or we need to solve if it gets into some kind of error.

What is your hardware setup exactly? A headless PC with two SATA disks?

trumfnator, 2024-04-28:
@openoms here is the output:

[image]

setup is a x220 laptop. WWAN module replaced with m.2-ssd as OS disk and a 1tb ssd in the regular hard drive slot of the laptop.

trumfnator, 2024-04-28:
Nevermind, I could kill the process in the end and could use apt-get on my own, but i couldn't install btrfs-progs:
[image]

trumfnator, 2024-04-28:
Okay, I will close this issue.

Source of problem was a DNS Server issue on my part.

closed 2024-04-28 by trumfnator

openoms, 2024-04-29:
Great, thank you for documenting the issue.

rootzoll, 2024-12-17:
@314159-r try latest v1.11.3 image - if I remember correctly it checks for BTRFS now a different way.

314159-r, 2024-12-19:
Yes, that worked! 
Upgrading from v1.10.0 to v1.11.3 completed without issues. I'm up and running... 

And thank you for responding to this old closed issue. I hesitated to post here, but the the issue description by @trumfnator matched mine perfectly, although for another reason.

314159-r, 2024-12-19:
...well, the upgrade completed... but it wiped my LNbits setup clean!
All users, wallets and settings are gone. 

Luckily for me, I had no funding in there. So the only harm is the loss of some experimental configuration. Instead I get a new opportunity to explore LNbits. For example: Learn how to backup...

I assume this is caused by me skipping versions, and jumping straight to v1.11.3. LNbis appear to have got some substantial development lately. _(In hindsight, I should probably have opted for a minimal jump in version, and start to try with v1.11.1, v1.11.2 and so on, coming from v1.10.0)_.

Or -  Possibly, during my first upgrading attempt, my SD-card died. The node had been unstable some week before, which triggered my upgrade. I had to flash a new card with a fresh v1.10.0, and use that to initiate the upgrade. 
And - I haven't used my LNbits for probably half a year, so I don't really know when it happened...