raspiblitz/raspiblitz #4559: Boot Process Gets Stuck

rwernerar, 2024-04-27:
I have attempted a fresh, clean installation of the latest version. However, there seems to be absolutely no way to make it happen! The boot process gets stuck. The only remaining option to try is changing the SD card. Nevertheless, the same card works perfectly when flashing Raspberry Pi OS.

I'm running a Raspberry Pi 4 with 8 GB, connected to an HDMI screen. If anyone has any suggestions, I would be grateful. I'll keep trying regardless!
[image]

rwernerar, 2024-04-29:
UPDATE: 
I've purchased a new SanDisk EXTREME PRO 64GB microSD card. I flashed the brand-new card with the latest versions of Raspiblitz and Balena. However, the boot process gets stuck at the exact same place as before.

For reference, I've attempted to fix the filesystem using both Linux and Windows. In both cases, no errors are detected, and no issues are resolved. Additionally, I tried flashing the Raspberry OS for Raspberry Pi 4, and that worked without any issues.

rwernerar, 2024-05-26:
Yes, I have tried that, and in fact, no errors were found (for both: the SSD and microSD card). What's even more, I erased the whole SSD disk and reformatted it. Always the same issue. The fsck is being run on the microSD card. The hard disk is a SAMSUNG 870 EVO SATA III SSD 1TB 2.5" (not NVMe)

The solution to this problem was to download the previous version of RaspiBlitz, flash the microSD card, set up everything, and then update to the newer version. I have no idea why it works this way!

rootzoll, 2024-05-27:
OK strange - but good that you found a way to resolve it. 

I will close the issue - but if anybody has the same error please report & reopen.

closed 2024-05-27 by rootzoll

ferdinandhodl, 2024-07-14:
same issue on Raspberry 5 8GB

ferdinandhodl, 2024-07-14:
I managed to get it running by running `fsck -fy` on both partitions and thereby fixing the e2fsck error **before** booting the SD card for the first time.

rwernerar, 2024-07-14:
I can't remember now if I also tried to run fsck before booting the SD card for the first time, but in any case, I will take that option into account next time. 
Thanks.

cahueya, 2024-09-07:
I had the same issue with 1.11.2 after a unsuccessful update process. The only way out was to reflash the SD card after fully deleting (overwrite with zeros) and checking with fsck. I also ran the fsck on the SSD, but it reported fine.

mzander, 2024-12-05:
I have the same issue with several SD cards. I tried the fat and the min image.
I am now going to try version 1.11.1 of the min image.

philihp, 2025-07-08:
I was encountering this on a setup with formerly working hardware. It was like 1.08, so I thought I'd just wipe it and start a new install. It happened on the formerly working SD card as well as a [PNY 32GB Elite Class 10 U1 microSDHC](https://www.amazon.com/dp/B07R8GVGN9). I also tried swapping my old Raspberry Pi 4 with a new Pi 5.

What fixed it for me was installing with the _raspiblitz-min-v1.11.4-2024-12-24.img.gz_ image, rather than the fat one.