raspiblitz/raspiblitz #4565: bitcoind.service not working?

inpharmaticist, 2024-05-04:
**Describe the bug**
A clear and concise description about the issue you encounter.
Tried on v1.10 and v1.11. I can start bitcoind with

```
sudo bitcoind
```

but not 'bitcoind'

```
admin@192.168.1.67:~ ₿ bitcoind
Error: Settings file could not be read:
- /mnt/hdd/bitcoin/settings.json. Please check permissions.
```

nor can I get the systemd service to start. 

```
admin@192.168.1.67:~ ₿ sudo systemctl start bitcoind.service
Job for bitcoind.service failed because the control process exited with error code.
See "systemctl status bitcoind.service" and "journalctl -xe" for details.

```

When using 'systemctl status bitcoind.service' I get
```
bitcoind.service - Bitcoin daemon on mainnet
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-05-04 23:58:15 BST; 10s ago
    Process: 1813457 ExecStartPre=/home/admin/config.scripts/bitcoin.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 1813481 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 248ms
```

**To Reproduce**
Steps to reproduce the behavior.
I tried flashing a new version and it didn't create a symlink to the HDD. It was local. I created a symlink to the HDD, and can now start as described above.

**Debug file**

If possible, please add a debug report. To do that, drag a text file with the debug report in here.
https://termbin.com/0mus

rootzoll, 2024-05-24:
If you want to start bitcoin by direct command it needs to be with user `bitcoin`.. so it would be:
`sudo -u bitcoin bitcoind`

But I would need to look at the debug like to maybe find why your bitcoind.service is exiting - can you repost it, because I get a 404 on the link.

inpharmaticist, 2024-06-30:
It kept crashing, started a new node, will make a new issue.

closed 2024-06-30 by inpharmaticist