raspiblitz/raspiblitz #4567: [1.11.0 amd64] FAIL - npm run build did not run correctly, aborting (1)

adevoss, 2024-05-14:
```
mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.5.0
mempool revision 57eddac7f
src/resources/config.js file not found, creating new config file
CONFIG:  (function (window) {
  window.__env = window.__env || {};
    window.__env.GIT_COMMIT_HASH = '57eddac7f';
    window.__env.PACKAGE_JSON_VERSION = '2.5.0';
  }(this));
src/resources/config.js file saved

> mempool-frontend@2.5.0 ng
> ./node_modules/@angular/cli/bin/ng.js build --configuration production --localize

⠙ Generating browser application bundles (phase: sealing)...Killed
FAIL - npm run build did not run correctly, aborting (1)

```
**To Reproduce**
Install Debian 12 (no GUI)
Install `sudo`
Execute `build_sdcard.sh` (headless, no wifi)

adevoss, 2024-05-16:
Happens only with fatpack option.

rootzoll, 2024-10-01:
I tested fatpack on x86 VM .. stopped at Joinmarket - @openoms can you take a look?
```
Empfange Objekte: 100% (13601/13601), 9.70 MiB | 11.51 MiB/s, fertig.
Löse Unterschiede auf: 100% (8953/8953), fertig.
HEAD ist jetzt bei f4c2b1b Merge JoinMarket-Org/joinmarket-clientserver#1671: Release notes and version for v0.9.11
--2024-10-01 20:36:28--  https://github.com/kristapsk.gpg
Auflösen des Hostnamens github.com (github.com)… 140.82.121.3
Verbindungsaufbau zu github.com (github.com)|140.82.121.3|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 7834 (7,7K) [text/plain]
Wird in »pgp_keys.asc« gespeichert.

pgp_keys.asc                          100%[=======================================================================>]   7,65K  --.-KB/s    in 0,001s

2024-10-01 20:36:29 (12,3 MB/s) - »pgp_keys.asc« gespeichert [7834/7834]

pub   rsa2048 2017-05-10 [SC]
      C488D91AAA047BE8FDC4499CFF6564B611D533AF
uid                      Kristaps Kaupe <kristaps.k@intelligentsystems.lv>
sub   rsa2048 2017-05-10 [E]

pub   rsa4096 2019-07-06 [SC] [verfallen: 2022-06-07]
      BF60DF964F88DD88174089A2D47B1B4232B55437
uid                      Kristaps Kaupe <kristaps@blogiem.lv>

pub   rsa4096 2021-01-09 [SC]
      70A1D47DD44F59DF8B22244333E472FE870C7E5D
uid                      Kristaps Kaupe <kristaps@blogiem.lv>
sub   rsa4096 2021-01-09 [E]

gpg: Schlüssel FF6564B611D533AF: Öffentlicher Schlüssel "Kristaps Kaupe <kristaps.k@intelligentsystems.lv>" importiert
gpg: Schlüssel D47B1B4232B55437: Öffentlicher Schlüssel "Kristaps Kaupe <kristaps@blogiem.lv>" importiert
gpg: Schlüssel 33E472FE870C7E5D: Öffentlicher Schlüssel "Kristaps Kaupe <kristaps@blogiem.lv>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 3
gpg:                              importiert: 3
# goodSignature(0)
# correctKey(1)

# BUILD FAILED --> PGP verification not OK / signature(0) verify(1)
 Failed to install JoinMarket
```

openoms, 2024-10-02:
@rootzoll the env needs to be in English as we are checking for the "Good signature" in some scripts.

rootzoll, 2024-10-02:
@openoms  OK .. I will merg #4770 which should address this.

rootzoll, 2024-10-02:
OK tested the build script in fatpack for comming RaspiBlitz v1.11.3 on a debian12 AMD64 VM and it runs thru.

Closing for RaspiBlitz v1.11.3rc1 release.

closed 2024-10-02 by rootzoll