raspiblitz/raspiblitz #4568: [1.11.0 amd64] LNbits - Bad Gateway

adevoss, 2024-05-16:
After installing v1.11.0 on amd64 VM LNbits is not shown in browser.

**To Reproduce**
Install v1.11.0 with existing data drive from v1.10.0.

**Debug file**
https://termbin.com/w5ol

bvolution, 2024-05-16:
I've got the same issue on "fresh" VM with raspiblitz. MIgrated data from an old raspiblitz but with no LNBits installed.
I noticed several errors during the first time I installed LNBits, especially poetry failed to install one dependency.
This appears to also be visible in the `journalctl`

Here is the install log after I uninstall via menu -> service and reinstalled again (I notived however that poetry install is not run again). I will check the `bonus.lnbits.sh` tomorrow to find where LNBit is installed and if I can fix the poetry error.

PS: I also tried `sudo systemctl restrat lnbits.service`

## Journalctl lnbits

`sudo journalctl -u lnbits -n30`

```
Mai 17 00:12:19 raspiblitz-core sh[133331]: Run `poetry install` to resolve and get rid of this message.
Mai 17 00:12:19 raspiblitz-core sh[133331]: Traceback (most recent call last):
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<string>", line 1, in <module>
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
Mai 17 00:12:19 raspiblitz-core sh[133331]:     return _bootstrap._gcd_import(name[level:], package, level)
Mai 17 00:12:19 raspiblitz-core sh[133331]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
Mai 17 00:12:19 raspiblitz-core sh[133331]:   File "/home/lnbits/lnbits/lnbits/server.py", line 6, in <module>
```

## Error During Reinstall of LNBits via Menu

```
# Running: 'bonus.lnbits.sh on lnd'
lnbits
/usr/local/bin/poetry
# preparing env file
# install database: PostgreSQL
# Using the default PostgreSQL version: 15
# Install PostgreSQL
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
postgresql-15 ist schon die neueste Version (15.7-1.pgdg120+1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
# There is old data for 15, restoring ...
Assertion failed on job for postgresql@13-main.service.
**Error: cluster configuration already exists**
Synchronizing state of postgresql.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable postgresql
psql (PostgreSQL) 15.7 (Debian 15.7-1.pgdg120+1)
# wait for the postgresql server to start
Setting default password for postgres user
ALTER ROLE
OK PostgreSQL installed
# Generate the database lnbits_db
CREATE DATABASE
CREATE ROLE
GRANT
# Setup PostgreSQL successful, new database found:  lnbits_db
# Running: 'bonus.lnbits.sh switch lnd'
## bonus.lnbits.sh switch lnd
##############
# NOTE: If you switch the funding source of a running LNbits instance all sub account will keep balance.
# Make sure that the new funding source has enough sats to cover the LNbits bookeeping of sub accounts.
##############
# adding lnbits user is member of lndreadonly, lndinvoice, lndadmin
# preparing lnbits config for lnd
##############
# OK new funding source set - does need restart or call: sudo systemctl restart lnbits
##############

*** Updating Firewall ***
Rule added
Rule added (v6)
Rule added
Rule added (v6)

*** Install systemd ***
Created symlink /etc/systemd/system/multi-user.target.wants/lnbits.service → /etc/systemd/system/lnbits.service.
# OK - lnbits service is enabled, system is on ready so starting lnbits service
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful

# Hidden Service for lnbits
HiddenServiceDir /mnt/hdd/tor/lnbits
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:5002
HiddenServicePort 443 127.0.0.1:5003

Reloading Tor to activate the Hidden Service...

The Tor Hidden Service address for lnbits is:
****CENSORED*******
use with the port: 80
or the port: 443
# OK install done ... might need to restart or call: sudo systemctl start lnbits
result='OK'
```

rootzoll, 2024-10-01:
Finale tetsing with PR #4568

PR #4768 (2024-10-04): #4568 LNbits install fix for AMD & VMs + update lnbits to v0.12.11
fixes #4568 on my test machine & should also address #4765
closed 2024-10-04 by rootzoll