raspiblitz/raspiblitz #4569: [1.11.0 amd64] WebUI only shows Settings, Home and Apps show spinning circle

adevoss, 2024-05-16:
After provisioning and syncing blockchain Home page and Apps page show spinning circle.
Apps are installed.

**To Reproduce**
Install Debian 12
Execute build_sdcard.sh and reboot with existing datadrive.

**Debug file**
https://termbin.com/yfw8

FragoriK, 2024-05-19:
My node is up and running since 2 weeks and I have had this issue since scratch. The app page never loads and has the loading circle.
The home page sometimes load after a while but the « connection details » part , this one only loaded once and then never again.
Same issue on iphone or mac, also same issue with the last 2 releases.

FragoriK, 2024-05-19:
Also on the settings page, the raspiblitz version and the blitz-api don t show. Only the raspiblitz webui shows version 1.3.0-dev

velgard1, 2024-06-20:
I experience more or less the same issue - I think I had the same with V1.10 already, but it worked from time to time or took only very long to show up...

I don't see SAT/BTC values and a loading section, at the bottom in the Lightning section I also don't have values. It also shows only webui 1.2.0.

Apart from the UI everything seems to be fine, no matter if it's SSH, BitBanana (former ZAP) or RTL.

velgard1, 2024-06-23:
Reinstalled API+WebUI and it works now (with my Pi4 config V1.2.0 was installed).

woiwoiwoiwoi, 2024-08-05:
Same issue here. For Apps, I get a spinning circle. Uninstalling and reinstalling did not fix the issue.

adevoss, 2024-09-02:
Today I tried `build_sdcard.sh` from `dev` branch.
All fine accept LNbits (502 Bad Gateway)

StartmeupLuca, 2024-09-20:
> Used to have the same issue for months… then this morning I uninstalled and reinstalled webUI via ssh and now it works fine and smooth… Seems like Velgard1 did the same thing a month ago and has the same result, so maybe it just needs a uninstall / reinstall … fyi it can be done without restarting the node and takes only a couple minutes :)

how do you  uninstall and reinstall the webUI?

cstenglein, 2024-09-23:
> how do you uninstall and reinstall the webUI?

Via SSH:

Settings -> Uncheck `Blitz API + WebUI` and confirm to uninstall. Then re-check it and confirm to install.

cstenglein, 2024-09-30:
Could be fixed by https://github.com/fusion44/blitz_api/pull/272

rootzoll, 2024-10-01:
This issue got removed from v1.11.3 milestone because its collected as same/similar under: https://github.com/raspiblitz/raspiblitz/issues/3608

closed 2024-10-01 by cstenglein